MarlinFirmware/Marlin #20487: [BUG] G34 not achieving expected accuracy and inconsistency between docs and expected accuracy

bakaufman, 2020-12-16:
### Bug Description

Something isn't right with G34. It doesn't achieve the level of accuracy claimed.

#define Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS
  #if ENABLED(Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS)
    // Define Stepper XY positions for Z1, Z2, Z3 corresponding to
    // the Z screw positions in the bed carriage.
    // Define one position per Z stepper in stepper driver order.
    #define Z_STEPPER_ALIGN_STEPPER_XY { { -15, 280 }, { -15, 25 }, { 290+46, 154 } }//BAK reversed for coreYX
  #else
    // Amplification factor. Used to scale the correction step up or down in case
    // the stepper (spindle) position is farther out than the test point.
    #define Z_STEPPER_ALIGN_AMP 1.0       // Use a value > 1.0 NOTE: This may cause instability!
  #endif

  // On a 300mm bed a 5% grade would give a misalignment of ~1.5cm
  #define G34_MAX_GRADE              5    // (%) Maximum incline that G34 will handle
  #define Z_STEPPER_ALIGN_ITERATIONS 5    // Number of iterations to apply during alignment
  #define Z_STEPPER_ALIGN_ACC        0.001 // Stop iterating early if the accuracy is better than this
  #define RESTORE_LEVELING_AFTER_G34      // Restore leveling after G34 is done?
  // After G34, re-home Z (G28 Z) or just calculate it from the last probe heights?
  // Re-homing might be more precise in reproducing the actual 'G28 Z' homing height, especially on an uneven bed.
  #define HOME_AFTER_G34 //BAK
when I run G34 i5 I get this:
Recv: G34 Iteration: 1
[...]
Recv: PROBEPT_0: 11.34
Recv: PROBEPT_1: 11.34
Recv: PROBEPT_2: 11.34
Recv: CALCULATED STEPPER POSITIONS: Z1=11.34 Z2=11.33 Z3=11.35
Recv: 
Recv: DIFFERENCE Z1-Z2=0.00 Z2-Z3=0.01 Z3-Z1=0.01
Recv: 
Recv: G34 Iteration: 2
[...]
Recv: PROBEPT_0: 11.35
Recv: PROBEPT_1: 11.34
Recv: PROBEPT_2: 11.35
Recv: CALCULATED STEPPER POSITIONS: Z1=11.35 Z2=11.34 Z3=11.35
Recv: 
Recv: DIFFERENCE Z1-Z2=0.01 Z2-Z3=0.00 Z3-Z1=0.00
Recv: Target accuracy achieved.
Recv: Did 2 of 5
Recv: Accuracy: 0.01
Recv: echo:Active Extruder: 0
[...]
Recv: echo:Active Extruder: 0
Recv: X:130.00 Y:82.00 Z:10.00 E:0.00 Count A:42527 B:-9629 Z:16000
Recv: echo:Active Extruder: 0
However, when I probe those 4 points that were tested, I get this: 
X    y    z    T0.010    T0.010
2    280    -4.12    -4.13    -4.14
2    51    -4.13    -4.13    -4.13
288    154    -4.08    -4.07    -4.08
The right point is low by 0.05 or more.
This is consistent with what I get for a 4 point bed leveling: 
Recv: Bilinear Leveling Grid:
Recv:       0      1
Recv:  0 -4.043 -4.130
Recv:  1 -3.954 -3.967
which is visualized as such:

I have done G28, G34, G29 multiple times and the data are very reproducible. the G29 points are equidistant to bed magnets to eliminate influence. I think that either my settings for the G34 in configuration_adv are wrong, or there is something wrong with the calculations.
Lastly, the precision allowed in the firmware is supposed to be "Target accuracy - must be between 0.001 - 1.0" https://marlinfw.org/docs/gcode/G034.html,  but in actuality, even if you set 
#define Z_STEPPER_ALIGN_ACC        0.001
 you cannot call for G34 i8 T0.005, you just get an error saying the limit is 0.01

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Set up G34 for T0.001
2. execute G34 i5 T0.01 (works) or T0.001 (doesn't work)
3. Test level with either 4 point bilinear or the exact points measured by G34. Right side is always much lower than expected

**Expected behavior:**

I expect G34 i5 T0.01 to achieve less than 0.01 deviation between Z1, Z2, Z3
I expect to be able to perform G34 i8 T0.001 as per the MARLIN G34 description online

**Actual behavior:**

Demonstrable that G34 i5 T0.01 achieve 0.05-0.06 at the points probed. Is there a rounding error somewhere? 
G34 i5 is limited to T0.01 in execution, although you can compile with 0.001 defined in firmware, just not execute it.

#### Additional Information
Please see attached
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5706044/Archive.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

bakaufman, 2020-12-17:
I was able to fix some of this problem by defining the G34 grid positions manually to avoid magnets. This solved the inconsistency between the 3 points. It doesn't explain the potential documentary error between T0.01 and T0.001. Let me know how to improve that language so it will all be 0.01 tolerance and we can move on.

thisiskeithb, 2020-12-17:
> It doesn't explain the potential documentary error between T0.01 and T0.001. Let me know how to improve that language so it will all be 0.01 tolerance and we can move on.

You can open a new issue in the Marlin Documentation repo (or create a PR) to update G34 documentation: https://github.com/MarlinFirmware/MarlinDocumentation/issues

> I was able to fix some of this problem by defining the G34 grid positions manually to avoid magnets. This solved the inconsistency between the 3 points.

Closing since this was not a Marlin bug.

closed 2020-12-17 by thisiskeithb