MarlinFirmware/Marlin #20488: [BUG] FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH

CRCinAU, 2020-12-17:
Configurations:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/CR-10%20V2

When compiling with a filament sensor, the following error is presented:
```
FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH
```

In the example config files, the settings are:
```
#define EXTRUDE_MAXLENGTH 435 (Configuration.h)
#define FILAMENT_CHANGE_UNLOAD_LENGTH      435 (Configuration_adv.h)
```

As such, the settings do meet the error message requirements.

Likely where this is set, the sanity check is `<` instead of `<=`

ellensp, 2020-12-17:
yes it does do this incorrectly. The test is
`#elif ENABLED(PREVENT_LENGTHY_EXTRUDE) && FILAMENT_CHANGE_UNLOAD_LENGTH > EXTRUDE_MAXLENGTH`

thisiskeithb, 2020-12-17:
> yes it does do this incorrectly. The test is
> `#elif ENABLED(PREVENT_LENGTHY_EXTRUDE) && FILAMENT_CHANGE_UNLOAD_LENGTH > EXTRUDE_MAXLENGTH`

That looks correct.

I grabbed `bugfix-2.0.x` from 6d47baee5dfcdfc7b8753eb76f4fb01be2fb5d4a and CR-10 V2 configs from [`6150f12`](https://github.com/MarlinFirmware/Configurations/tree/6150f12b34609788568ad656e9b4258cc4497e18) (due to issues with homing feedrates not working at the moment) and they compiled fine in VSCode + PlatformIO.

From `Configuration.h`:
```cpp
#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 435
```
From `Configuration_adv.h`:
```cpp
#define FILAMENT_CHANGE_UNLOAD_LENGTH      435
```

Reducing `EXTRUDE_MAXLENGTH` to 434 causes the `FILAMENT_CHANGE_UNLOAD_LENGTH must be less than or equal to EXTRUDE_MAXLENGTH` sanity check to kick in.

I can't reproduce the error.

ellensp, 2020-12-17:
After sleep, my brain is working a little better, and I also cannot reproduce this error.  closing

closed 2020-12-17 by ellensp

CRCinAU, 2020-12-17:
You're right - after getting at least 5 hours of contiguous sleep in one day, I spotted the error. This was indeed a PEBKAC.

Apologies for the noise.