MarlinFirmware/Marlin #20494: [BUG] (2.0.7.2 temp bug)

gordonf130, 2020-12-17:
### Bug Description
iUpdated from 2.0.5 to 2.0.7.2 everything was working fine, now i have a problem with hot end temp going way above what I set, and it's not overshooting as it goes 100c + and never comes back down, just keeps climbing. i use a e3d v6 hotend and select thermocouple 5, the same one I used before and worked fine then. when i pid tune the hot end it holds temp fine at 200 then update the settings tried with terminal and in firmware, then reset epprom, then when i set temp to 100c it climes over 180c, if i set 200c it climes over 275 and triggers thermal runway  

### Configuration Files

[marlin.gordon.zip](https://github.com/MarlinFirmware/Marlin/files/5711339/marlin.gordon.zip)

### Steps to Reproduce

1. set up Malin 2.0.7.2 with my printers settings using e3d hotend tmc2209 drivers, 
2. compile the firmware and upload it to the skr1,3 board with micro sd.
3. set hotend to 100c to test hot end was heating up okay and getting to set temperatures to make sure all setting are wright and it goes way past and keeps climbing 

**Expected behavior:**
the printer should hold at set temperature  like it dose when pid tunning,  

**Actual behavior:**
when setting the temperature manually it keeps climbing past what was set and keeps going till it hits max temp
doing a pid tune on the hot end it keeps the temp at the temp in the pid settings
starting a print and letting it heat up will also hit thermal run away 

#### Additional Information
[image]
[image]

* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

boelle, 2020-12-17:
> iUpdated from 2.0.5 to 2.0.7.2

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

gordonf130, 2020-12-17:
sorry have tested with the 2.0.x bugfix and it is doing the same thing, was all working fine with 2.0.5

gordonf130, 2020-12-17:
[marlin,bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/5711898/marlin.bugfix.zip)
this is my congif and configadv from the latest 2.0.bugfix

ellensp, 2020-12-17:
I don't think PID_OPENLOOP does what you think it does.
Disable this

gordonf130, 2020-12-18:
> I don't think PID_OPENLOOP does what you think it does.
> Disable this

trying now will report back in 5

gordonf130, 2020-12-18:
> I don't think PID_OPENLOOP does what you think it does.
> Disable this

thank you very much, this is not a bug after all I just looked it many many many times, you have solved hours upon hours of trying different setting, changing thermistor config, hours of googling and scrolling tho pages looking for answers, I honestly can't thank you enough I was at the point of pulling my hair our, I will close this now/ honestly mate lifesaver

closed 2020-12-18 by gordonf130