MarlinFirmware/Marlin #20496: [BUG] BL-Touch Leveling Causes Stepper Jam

omarhabib11, 2020-12-17:
Bug Description
Hello. The Creality CR-10 V2 configuration with BL-Touch fails to probe successfully. It will probe as expected along the first row, but when it gets to the second row along the Y axis, it proceeds to probe the same spot multiple times. The printer reports the X axis as moving properly for the subsequent probes, although the probe stays in the same spot, until it moves down another row along the Y axis, thinking it is at the reported X axis. It thinks it is near X=0, but actually, X is almost at the end because it never moved backwards, so it tries to move past the end of the X axis for the next probes, causing the stepper to jam and make an awful noise.

See: [https://github.com/MarlinFirmware/Configurations/issues/365]

Configuration Files
This bug was seen on default configuration, although I have tinkered with different settings such as UBL with no luck.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5711975/config.zip)

Steps to Reproduce
Start Bed Levelling
Wait for failure around probe 6 or 7
Expected behavior:

Probing should not cause the stepper to jam.

Actual behavior:

Stepper jams on the X axis after a few probes.

thisiskeithb, 2020-12-17:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/365

thisiskeithb, 2020-12-17:
Please don't create multiple reports for the same issue.

closed 2020-12-17 by thisiskeithb

omarhabib11, 2020-12-17:
Sorry, it seems like the issue lies deeper than configuration, so I wanted to move it to the main issues page.

thisiskeithb, 2020-12-17:
@sjasonsmith / @rhapsodyv Can you transfer https://github.com/MarlinFirmware/Configurations/issues/365 here?

sjasonsmith, 2020-12-18:
Duplicate of #20497