MarlinFirmware/Marlin #20497: BL-Touch Leveling Causes Stepper Jam

omarhabib11, 2020-12-17:
### Bug Description

Hello. The Creality CR-10 V2 configuration with BL-Touch fails to probe successfully. It will probe as expected along the first row, but when it gets to the second row along the Y axis, it proceeds to probe the same spot multiple times. The printer reports the X axis as moving properly for the subsequent probes, although the probe stays in the same spot, until it moves down another row along the Y axis, thinking it is at the reported X axis. It thinks it is near X=0, but actually, X is almost at the end because it never moved backwards, so it tries to move past the end of the X axis for the next probes, causing the stepper to jam and make an awful noise.

### Configuration Files

This bug was seen on default configuration, although I have tinkered with different settings such as UBL with no luck.

[config.zip](https://github.com/MarlinFirmware/Configurations/files/5671485/config.zip)

### Steps to Reproduce

1. Start Bed Levelling
2. Wait for failure around probe 6 or 7

**Expected behavior:**

Probing should not cause the stepper to jam.

**Actual behavior:**

Stepper jams on the X axis after a few probes.

sjasonsmith, 2020-12-18:
Can one of you who tested with bugfix-2.0.x please post your configuration files that compile with the bugfix branch?

dirk-schulz, 2020-12-18:
Hi Jason Smith,
yes, sure. ;-)

Do you think it is an issue in the config?
Sometimes I get an error message during probe saying something like "printer issue - temperature sensor" or "printer unkown issue during probing". The message comes as fast as it went away, so I could not clearly remember the exact error message. Does marlin somewhere writes a log, where I may can review the error message?
But mostly the error occurs like described 5 day ago - so without error message and only not moving the head correctly.
For me as no-prgrammer it looks like the program runs in an error while executing the probe which results in reseting the current x/y/z values. So maybe more a bug in the code?

Here my used config file. 
Looking forward for your feedback and thanks in advance.
[Marlin_config_DS_with_bugfix-2.0.x_Sidewinder_X1.zip](https://github.com/MarlinFirmware/Marlin/files/5716836/Marlin_config_DS_with_bugfix-2.0.x_Sidewinder_X1.zip)

dirk-schulz, 2020-12-28:
Hope you had merry christmas days.
Are there any news about the bug?

Lukapp, 2021-01-18:
I can confirm that it happens to me in an Artillery Genius from versions 2.0.5, now in 2.0.6 too with ABL, and I see that it also happens in 2.0.7.2. It always happens on the X axis at random, almost always in the first row (4x4 or 5x5) or other times in any other row. And I had tried to do PROBE_Y_FIRST and it also happened, but in the Y axis. I have also observed in my case it happens when start a print, if I do the ABL first it does not happen. So at the moment I do an auto-leveling first, and in the start gcode I replace G29 with M420 S1 so that it takes the last parameters. It always happens between points of X - X from right to left, or Y-Y if I define it PROBE_Y_FIRST. I always make an origin of the axes XY before printing, and also sometimes it stops earlier, as something resets the motors and it just didn't get to the endstops. Throughout the printing process, no problem.

francisduvivier, 2021-01-22:
Hi, seems that with me this issue had to do with endstop noise (even though my endstops seem to have capacitors) because I was able to fix it with by uncommenting `#define ENDSTOP_NOISE_THRESHOLD 2`
I see that this is still commented out in the attached zip, so might be worth trying out)

I did however also sometimes have issues with homing before uncommenting that, and those are also fixed by it, but I didn't think it would also affect the leveling moves.

closed 2021-03-01 by github-actions[bot]