MarlinFirmware/Marlin #20503: Probing fails on marlin 2.0 on SKR 1.4

kmichael, 2020-12-18:
I've build a custom delta with BTT SKR 1.4 as a brain with TMC2208 drivers. Recently I've added the Z Probe using nozzle as a probe and again custom designed head with microswitches as probe sensor. It was working correctly until recently it started failing the bed leveling.
The probe behaves very strangely, the symptoms are:
1. Probing with `G29 V4` - each consecutive probing point report lower Z value. And I mean much lower. In around 7 points we reach -5.9 when `Error: Probing Failed` gets reported. The nozzle then stops around 10-20mm above the bed and when jogging it down 0.1mm it actually goes up and says it's at Z=0.0. The bed is level, at least definitely more level. When it works, because it sometimes does, the Z values are around 0+/-0.4.
2. Similar issue started later occurring with `G33` command. Again after some points probing fails with nozzle apparently being at Z=-5.9.
3. Later it started to behave even more strange. Again, when issuing `G29 V4` the first point it goes to is `X0 Y-50` But actually it moves only a few mm in Y direction, probes there and then moves to the next point a few cm above the bed and again `Error: Probing Failed`. It's like the plane it tries to probe is twisted along X axis by 20-30 degrees.
4. When manually probing that first point with `G30 X0 Y-50` everything works fine. When I move around there with `G0 X0 Y-50 Z10` and then do the `G29` it works.

I've started experiencing it on some old `bugfix-2.0.x`. Updating it to the last commit didn't help. I tried `2.0.7` and newest version from `2.0.x` and it hasn't helped.

FanDjango, 2020-12-18:
Losing steps or 2208 to hot and cycling down (too hot) for short periods?

kmichael, 2020-12-18:
> Losing steps or 2208 to hot and cycling down (too hot) for short periods?

Hi, thanks for the quick reply. 2208 have radiators and a fan blowing on them. But even if it wasn't the case, leveling is the first step of the print, so it often happens right after I turn on the printer - the steppers are cold.

Still - loosing steps could cause slight differences, but that's definitely to much. I've suspected it anyway, but turning down the speed by 3-4 times changed nothing. Plus it prints normally, the issue only occurs when probing and looks like for some reason the printer doesn't know where it is. Is there maybe some flag to set, that it would report it's posision - or rather where it thinks it is? But do this during the movements or something? I'm not talking about `M114`

FanDjango, 2020-12-18:
Ok, how about reducing the situation to solely up/down movement first. M48, up down many times, see where it goes?

FanDjango, 2020-12-18:
> the issue only occurs when probing and looks like for some reason the printer doesn't know where it is

Ok, let's prove that

FanDjango, 2020-12-18:
>  Is there maybe some flag to set, that it would report it's posision

Not really, but turn on "debuging_levelling", see "configuration.h"

kmichael, 2020-12-18:
> > Is there maybe some flag to set, that it would report it's posision
> 
> Not really, but turn on "debuging_levelling", see "configuration.h"
> 
> ```
> /**
>  * Enable detailed logging of G28, G29, M48, etc.
>  * Turn on with the command 'M111 S32'.
>  * NOTE: Requires a lot of PROGMEM!
>  */
> //#define DEBUG_LEVELING_FEATURE
> ```
> 
> Will give you positions also.

Cool, will try it and let you know, but that's tomorrow - I'm in Europe and it's getting late here ;)

ellensp, 2020-12-18:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thierryzoller, 2020-12-21:
Try swapping the driiver with another axis and see whether you have the same issue.

sjasonsmith, 2020-12-22:
Closing, as this has insufficient information to act upon. In the future, please be sure to include all the information requested in the issue template when filing a report. This can be re-opened if you update the description to match the template @ellensp shared with you a few days ago.

closed 2020-12-22 by sjasonsmith