MarlinFirmware/Marlin #20505: [BUG] platformio init fails (using CLion) due to fysetc_f6_14 board

canihavesomecoffee, 2020-12-18:
### Bug Description

When cloning Marlin, and trying to set up Marlin with PlatformIO through CLion, it always fails with the next error:

```
/home/me/.local/bin/platformio init --ide clion

The current working directory /home/me/Documents/3D_printing/firmware_compilation/SKR-1.1-FFS will be used for the project.

The next files/directories have been created in /home/me/Documents/3D_printing/firmware_compilation/SKR-1.1-FFS
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File
Error: Unknown board ID 'fysetc_f6_14'

Process finished with exit code 1
```

### Configuration Files

N/A for this bug

### Steps to Reproduce

1. Clone Marlin, optionally check out to a tag
2. Open the project in CLion and let it initialize (using the `platformio init --ide clion`), or alternatively initialize it yourself through a terminal
3. Note that it errors with the error reported above

**Expected behavior:**

PlatformIO initializes correctly

**Actual behavior:**

PlatformIO errors out with "Unknown board ID 'fysetc_f6_14'"

#### Additional Information

If I comment out the section of the fysetc board, it works as intended, and then I can use CLion in combination with PlatformIO.

rhapsodyv, 2020-12-18:
Let's time I checked, CLion didn't work fine with PIO yet. Which version are you using?

canihavesomecoffee, 2020-12-19:
> Last time I checked, CLion didn't work fine with PIO yet. Which version are you using?

PIO is version 5.0.3, CLion is version CLion 2020.3 (CL-203.5981.166, built on December 1, 2020).

This issue has been persisting for a couple of months already, but last time I thought it was indeed related to the bad support of CLion. That has improved though (now you can select a PIO target for CMake builds and select the proper board whereas 2 months ago you had to do some trickery), but this error with the Unknown board ID remains unfortunately.

ellensp, 2020-12-19:
Platformio also doesn't know about this board
I you set motherboard BOARD_FYSETC_F6_14 and default_envs = FYSETC_F6_14
you get this error
```
> Executing task: pio run <

Processing FYSETC_F6_14 (platform: atmelavr; board: fysetc_f6_14; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown board ID 'fysetc_f6_14'
The terminal process "pio 'run'" terminated with exit code: 1.
```

rhapsodyv, 2020-12-19:
It's not a Marlin bug. PIO don't ship with this board. It has even a PR opened for that here: https://github.com/platformio/platform-atmelavr/pull/180

But we can copy that board file to marlin `buildroot/share/PlatformIO/boards` folder. It may solve both issues.
`https://raw.githubusercontent.com/platformio/platform-atmelavr/200d0e0150094286dc6b7bf9ebb3551ae4edf8bc/boards/fysetc_f6_14.json`

rhapsodyv, 2020-12-19:
This should fix it: #20507

canihavesomecoffee, 2020-12-19:
> It's not a Marlin bug. PIO don't ship with this board. It has even a PR opened for that here: [platformio/platform-atmelavr#180](https://github.com/platformio/platform-atmelavr/pull/180)
> 
> But we can copy that board file to marlin `buildroot/share/PlatformIO/boards` folder. It may solve both issues.
> `https://raw.githubusercontent.com/platformio/platform-atmelavr/200d0e0150094286dc6b7bf9ebb3551ae4edf8bc/boards/fysetc_f6_14.json`

I tried this out, and it does fix the issue indeed. Thank you for tracking this down and opening the PR :+1: 

Leaves me to wonder why no others encountered this before (although that can probably be explained because most will use PlatformIO through VS code, where this error doesn't surface for whatever reason)...

rhapsodyv, 2020-12-19:
> Leaves me to wonder why no others encountered this before (although that can probably be explained because most will use PlatformIO through VS code, where this error doesn't surface for whatever reason)...

Well.... when the fix is so simple (like just copying a file), some users that find it on forums and copy the file... or, no one are using that board :-p

canihavesomecoffee, 2020-12-19:
@rhapsodyv Just an FYI, I now encounter the same issue (being on the bugfix-2.0.x branch right now) with the `megamega1280` board, which was added/modified in #20493. I guess I'll download the JSON from here (https://github.com/platformio/platform-atmelavr/blob/7d3906c20d32751459e95fb74041e64f7a1160bf/boards/megaatmega1280.json), but this might be confusing for new users if they get missing boards...

rhapsodyv, 2020-12-20:
Sure, looks like a typo! I will fix. Thanks!

sjasonsmith, 2020-12-22:
It looks like that last item was fixed in https://github.com/MarlinFirmware/Marlin/commit/777c50a1ecd68c8b35531b654498e73bd5078b24

PR #20507 (2020-12-22): Add Missing FYSETC F6 14 Board
### Description

This board have a ENV on Marlin, but PIO doesn't have such board yet. There's a open PR for that, for more than one year. So I adding it to Marlin, to avoid breaking any builds.

Open PIO PR: platformio/platform-atmelavr#180

### Benefits

Fix #20505

### Related Issues

#20505
closed 2020-12-22 by sjasonsmith