MarlinFirmware/Marlin #20511: [FR] Support UART_ENDSTOPS (like SPI_ENDSTOPS but for TMC2209)

vedran-l, 2020-12-20:
I propose to support sensorless homing without connecting the DIAG pin of TMC2209 to limit pin and reading the stall status via UART instead, similarly like it's done for TMC2130 via SPI. TMC2209 datasheet indicates this could be done by reading SG_RESULT and comparing it to SGTHRS as the drive itself is doing exactly that to set DIAG pin.

Why would one want this? I'm putting together a machine with sensorless homing on a pin poor controller. In order to have enough pins, I resorted to remapping driver enable pins to UART in favor of software enabling the drivers and also using regular limit pins for other functions. When DIAG output on TMC2209 is currently used it must be connected to limit pin so it's can't be used otherwise. It also seems enable pin is needed to clear the latched DIAG state after limit is triggered.

LinFor, 2020-12-20:
I think uart interface too slow to do this...

vedran-l, 2020-12-20:
"Think" as in it's been tried and found lacking or just a feeling?

This is my thinking:
Homing @100mm/s with 80 steps per mm is 8k steps/s. Times three in case of a Delta or times two in case of a CoreXY since all moving drives need to be read, so 24k steps/s. It needs to read 10bit word so something like 240kbaud read plus whatever is needed to trigger a read.  Not doable for SW serial but possible in HW. However, drive only sets DIAG on each full step so divide everything by 16 (for microsteps) and we come down to 15kbaud which is doable even in SW.
Worst case, on a delta with multiplexed drivers over SW serial one would need to knock down homing speed a bit. On plain cartesian back of the envelope says it's possible.

If I'm wrong please do tell :)

sjasonsmith, 2020-12-20:
Reading a register requires a 4 byte request, a default 8 bit delay, and an 8 byte response. At 250kbaud this would be about 500 microseconds to read from a single driver.

sjasonsmith, 2020-12-20:
Can you describe what is your use-case? Why don't you just use a controller with more pins?

vedran-l, 2020-12-20:
I see, so it's ~6 times worse then my estimate... still doable, but getting quite marginal and probably not in SW.

Use case... well stupidity/stubbornness I guess would be most accurate :) I'm trying to use the board that I've got with the printer (Robin Nano 1.2 in Sapphire Plus). I added a second extruder, passively heated chamber+fan and temperature controlled extruder fans. I still have one pin left and WIFI port is still reserved for, well, WIFI :) That's the reason why I'm low on pins.
I'm trying to avoid switching to a different controller as this one is already nicely mounted inside the printer together with a FSMC LCD, and because a replacement costs money and time to ship from China.

Advantage to community would be ability to use sensorless homing on any board by hijakcing enable pin which is invariably both present and close by to wire into USART/PDN pin, with no rat's nest of wires that usually accompanies retrofitted sensorless homing.

thisiskeithb, 2020-12-20:
> still doable, but getting quite marginal and probably not in SW.

Sensorless homing can be difficult to tune in the best crimstances, so allowing an option that will likely lead to more support tickets probably isn't the best idea.

Having said that, we could add a little blurb in that section to warn users of possible unreliability like was done for sensorless probing with StallGuard if this FR were to move forward.

vedran-l, 2020-12-20:
I see, OK, I'll try something different. Thanks. Closing this FR

closed 2020-12-20 by vedran-l