MarlinFirmware/Marlin #20513: [FR] UNLOAD bed position after STOP command

mrx23dot, 2020-12-20:
### Description
After issuing STOP command it would be very useful to have the bed in UNLOAD position, not in HOME, in order to clean up.

### Feature Workflow
1, Printing starts
2, Printing fails by any means (eg. won't stick to bed)
3, I issue STOP via LCD
3.5, I don't think HOMING is necessary/good, STOP could be a critical mechanical fault (HOMING will happen on new print)
4, I would expect the bed to be in UNLOAD position so I can clean up the (non removable) bed 
    We already know the UNLOAD position because it's issued after print FINISH.
5, stepper motors should post-run (self-hold) for a few minutes to help scraping off the bed
6, point 5 should also happen after print FINISH to help removing the printed part.

#### Additional Information
I have ender3 XYZ in mind where the bed moves when I try to scrape it off, and home position makes bed inaccessible.

thisiskeithb, 2020-12-20:
You can already customize Marlin's actions when you abort a print. Just change `G28XY` to whatever you want:

https://github.com/MarlinFirmware/Marlin/blob/777c50a1ecd68c8b35531b654498e73bd5078b24/Marlin/Configuration_adv.h#L1206

closed 2020-12-20 by thisiskeithb

mrx23dot, 2020-12-20:
Thank you!
What's the finishing G-code to unload the bed? (Y-max for me)
I couldn't find near M1001 in code.