MarlinFirmware/Marlin #20516: [BUG] Resume print after power loss

akurekin, 2020-12-20:
### Bug Description

Using BIGTREETECH SKR MINI E3 V2.0 control board. Experiencing problem when I resume printing during power loss recovery.
When POWER_LOSS_RECOVERY is enabled, BACKUP_POWER_SUPPLY is enabled and POWER_LOSS_RECOVER_ZHOME is disabled the following abnormal behaviour is observed:
1. The printer is equipped with bigtreetech ups 24v v1.0. When I switch off the power the printer pauses printing, raises its print head and saves configuration parameters on an SD Card.
2. When I switch on the power again and select the resume printing option, the printer does homing of XY axes, heats the head bed and heats the head. Then it moves to the XY position where it stopped printing at power off. At the next stage it is about to resume printing. It starts lowering down the nozzle but goes too far crashing into the heat bed.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5720762/Configuration.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Switch off the power supply. Wait for printer to complete the power off procedure (raise nossle up, switch off)
2. Switch power on.
3. Select resume printing

**Expected behavior:**

When the printer powers on it is expected to: home XY axes; heat up the head bed and the head; move to the XY position where it stopped printing at power off; lower Z axes to the position when it stopped; resume printing.

**Actual behavior:**

Instead of returning back to its previous Z position when it stopped printing, the printer keeps lowering the nozzle until it crashes into the heat bed.

#### Additional Information

The wrong printer behaviour when it resumes printing can be explained by the following bug. When it starts power loss recovery it runs G92.9 command to set current nozzle position. The position supplied for Z axes is info.zraise . This command does not take into account home_offset and position_shift values. So, if the home_offset is not zero, the nozzle moves too far below the  position it occupied at the moment when the power was shut down.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

GoDotMad, 2020-12-20:
https://github.com/MarlinFirmware/Marlin/pull/20035#issue-516017686

this commit fix the problem 
it is already merged to bugfix2.0.x

so you can update to bugfix

or do as I did
comment this line (line 640 )
diveDir->close();
in file  /marlin/src/sd/cardreader.cpp

github051, 2020-12-21:
Did you try with Latest bug fix?

akurekin, 2020-12-21:
Tried with the latest bug fix, same problem, no difference!

github051, 2020-12-22:
I'm using power loss recovery without backup power and is working fine for me. 
Can you try without backup ?

akurekin, 2020-12-22:
Without backup it will work differently as it has no time to raise the nozzle up before switching off.
The solution with back-up that worked for me was to replace 
dtostrf(info.zraise, 1, 3, str_1)

with this:
dtostrf(info.zraise-home_offset.z-position_shift.z, 1, 3, str_1)

in the resume() function.

atharvbadkas, 2020-12-23:
I was also facing same problem...and opened same issue like this...but it was closed without solving the issue...hope this bug gets resolved.. really looking forward for it.
.
My [Issue](https://github.com/MarlinFirmware/Marlin/issues/20187) .
AND
I contacted @bigtreetech , they sent me Marlin firmware folder and bin file... claiming that they have tested it on SKR MINI E3V2 AND BTT UPS 24V for Ender 3 Pro but it doesn't work for me either...can you test this files and let me know your results???

[Drive Link](https://drive.google.com/folderview?id=1lLJ8wB38pCyJNlhxEXdSRXJNUCHG2Eis)

akurekin, 2020-12-23:
It is not the same issue. My issue is that the nozzle moves down too low when the printer resumes after power loss. Please see the suggested solution in my comment above.

atharvbadkas, 2020-12-24:
Do latest Bugfix works good for skr mini e3 v2 and Btt ups 24v??....i haven't tried it ince they closed the Issue...

akurekin, 2020-12-24:
I am using skr mini e3 v2 and Btt ups 24v and everything else works great.  It resumes printing after power loss.

theschles, 2020-12-29:
I'm using CopterTec build using Marlin 2.0.5.3 and it snapped off my BL Touch probe tip! Grrrr

Jedrusia, 2020-12-30:
> I am using skr mini e3 v2 and Btt ups 24v and everything else works great. It resumes printing after power loss.

Can you  attach Your all configuration files for working SKR MINI E3 V2 with BTT UPS 24V ???

atharvbadkas, 2020-12-30:
> I am using skr mini e3 v2 and Btt ups 24v and everything else works great. It resumes printing after power loss.

Can you send us your whole Marlin folder??.... that would be great

Jedrusia, 2020-12-30:
> I'm using CopterTec build using Marlin 2.0.5.3 and it snapped off my BL Touch probe tip! Grrrr

If your BTT UPS and SKR MINI E3 V2 mods work, it means that I either have a damaged UPS module or I cannot configure it to work, so I would like to ask for your configuration files, preferably the entire Marlin folder

atharvbadkas, 2021-01-03:
@akurekin Hi, Can you send us your marlin firmware folder which works for your setup??
Thanks

Jedrusia, 2021-01-03:
I think that is no option about correct working SKR MINI E3V2 with BTT UPS 24V 2.0. @akurekin not write the true. I was more corresponding with BTT and i was not make a positive result.
@akurekin 
If you write a true then please share a all config files.

akurekin, 2021-01-07:
Finally, here is the link on GD to the entire Marlin code that works for me. 
https://drive.google.com/file/d/10t5dIOadYNVE_aBKycG8hWd9363uEdwS/view?usp=sharing

I can confirm that this code works for SKR MINI E3V2 with BTT UPS 24V 2.0 
When I was testing my BTT UPS 24V 2.0 I came across a strange behaviour when power loss mode was triggered when I started printing and the printer was going into the halt state. I found the problem why that was happening and now everything works perfectly. The problem was not in the configuration of Marlin but actually it turn out to be  an electronics issue. The power off wire coming from BTT UPS 24V 2.0 to SKR MINI E3V2 wasaffected by the interferences that randomly triggered the Power OFF signal. I had to add an RC circuit to get rid of the interferences the Power OFF works as expected.

atharvbadkas, 2021-01-07:
> Finally, here is the link on GD to the entire Marlin code that works for me.
> https://drive.google.com/file/d/10t5dIOadYNVE_aBKycG8hWd9363uEdwS/view?usp=sharing
>
> I can confirm that this code works [...]

@akurekin Amazing, can you share images of your RC circuit and values you used of R and C??

ramsoni08, 2021-01-08:
> Finally, here is the link on GD to the entire Marlin code that works for me.
> https://drive.google.com/file/d/10t5dIOadYNVE_aBKycG8hWd9363uEdwS/view?usp=sharing
>
> I can confirm that this code works [...]

can you please share the picture of RC Circuit

Jedrusia, 2021-01-08:
> Finally, here is the link on GD to the entire Marlin code that works for me.
> https://drive.google.com/file/d/10t5dIOadYNVE_aBKycG8hWd9363uEdwS/view?usp=sharing
>
> I can confirm that this code works [...]

Please share a RC circuit and vaule of R and C elements

akurekin, 2021-01-08:
Sure, here is the circuit diagram:
[image]

Don't forget to print in LCD emulation mode from SD card inserted in control board and enable power loss recovery using gcode or fro LCD settings.

Good luck!

atharvbadkas, 2021-01-09:
> Sure, here is the circuit diagram:
> [image]
> 
> Don't forget to print in LCD emulation mode from SD card inserted in control board and enable power loss recovery using gcode or fro LCD settings.
> 
> Good luck!

Thank you very much
Will let you know what happens

thinkyhead, 2021-01-09:
>  I had to add an RC circuit to get rid of the interferences the Power OFF works as expected.

If this is solved then you can go ahead and close this issue.

atharvbadkas, 2021-01-09:
> > I had to add an RC circuit to get rid of the interferences the Power OFF works as expected.
> 
> If this is solved then you can go ahead and close this issue.

Hello,
Please wait until we confirm that this is the solution,
Let us try this on our machines, then close this issue.

atharvbadkas, 2021-01-09:
> Sure, here is the circuit diagram:
> [image]
> 
> Don't forget to print in LCD emulation mode from SD card inserted in control board and enable power loss recovery using gcode or fro LCD settings.
> 
> Good luck!

@akurekin  amazing work, it works as is should... thank you everyone 😀

akurekin, 2021-01-09:
Can I remind you guys that the discussion was started with the description of a bug in Marlin code that caused nozzle crash into a heat bed when the printing resumes after power off. I suggested a fix that works for me, but the main Marlin  code has to be updated to include the fix I suggested. Not sure if the issue can be closed.

ramsoni08, 2021-01-09:
> > Sure, here is the circuit diagram:
> > [image]
> > Don't forget to print in LCD emulation mode from SD card inserted in control board and enable power loss recovery using gcode or fro LCD settings.
> > Good luck!
> 
> Thank you very much
> Will let you know what happens

I printed a 3DBenchy for a test print. The printer showed me the message 3 times "Printer HALTED please reset" while printing and I have used this circuit in my mother board and UPS.

akurekin, 2021-01-09:
The capacitor should be soldered close to the control board and further away from the UPS

atharvbadkas, 2021-01-19:
> Can I remind you guys that the discussion was started with the description of a bug in Marlin code that caused nozzle crash into a heat bed when the printing resumes after power off. I suggested a [...]

I didn't saw behaviour like you explained in my setup, After resuming print head moves to previous location and starts printing. You might have enabled "Z homing after power loss".If this enabled then comment this line out.
@akurekin

github051, 2021-01-22:
> I didn't saw behaviour like you explained in my setup, After resuming print head moves to previous location and starts printing. You might have enabled "Z homing after power loss".If this enabled [...]

github051, 2021-01-22:
enabling "Z homing after power loss" X-gantry may hit your prints if the print size is bit big. Be careful

equisde, 2021-02-04:
i have an ender 3 pro v1.4 with the creality board and i'm not able to resume print also the resume menu is not shown i already enabled the powerloss feature on my marlin configuration any ideas?

github051, 2021-02-09:
> i have an ender 3 pro v1.4 with the creality board and i'm not able to resume print also the resume menu is not shown i already enabled the powerloss feature on my marlin configuration any ideas?

attach your config files and pin file

closed 2021-03-22 by github-actions[bot]