MarlinFirmware/Marlin #20520: [BUG] Cannot Compile Marlin 2.0.X on VSCode, even if using Auto Build when using env:STM32F103RET6_creality

Kamikazi615, 2020-12-20:
### Bug Description

Hi everyone, Im pretty new to all this but I truly tried everything I can do and my issue still persists.

Ive been following the exact steps of the following video https://www.youtube.com/watch?v=neS7lB7fCww&t=334s to configure my ender 3 v2's BL Touch and set its offset. However whenever it comes to compiling, I constantly get a "failure" despite following the exact steps in the video.

I figured that the issue occurs only when I try to use the env:STM32F103RET6_creality 

If i replace this with Mega2560, compiling works however I doubt I can use that in my printer.

As I said I'm very new to this so speak to me as a baby please haha!

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Using the lates Marlin 2.0.X Bugfix, try compiling when using the environment : STM32F103RET6_creality

**Expected behavior:**

Normally as seen in the video, VSCode should simply compile and then give me access to my firmware.bin

**Actual behavior:**

I get an error message stating:

Processing STM32F103RET6_creality (platform: ststm32@~10.0; board: genericSTM32F103RE; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (10.0.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 3.10000.201129 (1.0.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.99
|-- <Wire> 1.0
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduinoVariant/board.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduinoVariant/wirish/boards.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduinoVariant/wirish/start.S.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduinoVariant/wirish/start_c.c.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduinoVariant/wirish/syscalls.c.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/Print.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/WString.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/cxxabi-compat.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/ext_interrupts.cpp.o
Compiling .pio/build/STM32F103RET6_creality/FrameworkArduino/hooks.c.o
[... 118 lines cut ...]
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32F1/HAL.cpp.o] Error 1
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32F1/HAL_SPI.cpp.o] Error 1
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32F1/MarlinSerial.cpp.o] Error 1
*** [.pio/build/STM32F103RET6_creality/src/src/HAL/STM32F1/SPI.cpp.o] Error 1
================================================================ [FAILED] Took 9.57 seconds ================================================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:09.573
========================================================== 1 failed, 0 succeeded in 00:00:09.573 ==========================================================
The terminal process "pio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

#### Additional Information

I have tried contacting someone on Thingiverse and he has the exact same issue when trying to compile using  STM32F103RET6_creality. I have tried with Marlin 2.0.X and 2.0.6 and I keep getting the same issue. 

I have also tried using Marlin Auto Build, but it simply doesnt detect my STM32F103RET6_creality environment.

If I try doing all these modifications (the exact same as the video linked above) but with Mega2560 as an env, it compiles, so the problem really is the STM32F103RET6_creality environment.

Yes I am using Platform.io and opening my file correctly through it so that shouldn't be the issue.

Thanks for the help!

FanDjango, 2020-12-20:
This is not a bug. I suggest you go to discord ([marlin_discord](https://discord.com/invite/n5NJ59y). You'll get help there

thisiskeithb, 2020-12-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-20 by thisiskeithb

Kamikazi615, 2020-12-20:
Okay thanks for the info! Invite link for DC is invalid tho...

FanDjango, 2020-12-20:
Works for me. Try the one from @thisiskeithb below mine

Kamikazi615, 2020-12-20:
Both dont work... RIP

thisiskeithb, 2020-12-20:
They're the same link and both work for me.

You can join any Discord server manually by clicking on the + button below your list of servers & then click on the "Join a Server" button. Paste in the Marlin Discord server link from above and then click "Join Server".