MarlinFirmware/Marlin #20527: [BUG] MKS Robin Nano V2 timer conflict

trashypopashy, 2020-12-20:
### Bug Description

When trying to compile the latest Marlin bugfix 2.0.x, I am running into the Issue:
"Marlin\src\HAL\STM32\timers.cpp:281:41: error: static assertion failed: One or more timer conflict detected
  281 | static_assert(verify_no_duplicate_timers(), "One or more timer conflict detected")"

This happens, when using "default_envs = mks_robin_nano35_stm32" in platformio.ini(because i have a Robin Nano V2). When I try compiling with "default_envs = mks_robin_nano35", it shows no errors.

I have not tried to flash the compilable firmware with "default_envs = mks_robin_nano35", as I am afraid of bircking the board.

### Configuration Files

Did not find a possibility to attach files, here.

My configurations-files, I uploaded have many changes, but the behaviour is reproducable with only the follofing steps, as I guess:

### Steps to Reproduce

1. put in platformio.ini "default_envs = mks_robin_nano35_stm32"
2. put in configuration.h "#define MOTHERBOARD BOARD_MKS_ROBIN_NANO_V2", "#define SERIAL_PORT 3" and "#define SERIAL_PORT_2 1"

**Expected behavior:**

Should compile fine.

**Actual behavior:**

shows error: "Marlin\src\HAL\STM32\timers.cpp:281:41: error: static assertion failed: One or more timer conflict detected
  281 | static_assert(verify_no_duplicate_timers(), "One or more timer conflict detected")"

trashypopashy, 2020-12-20:
Here's the .zip
[config-files.zip](https://github.com/MarlinFirmware/Marlin/files/5721694/config-files.zip)

rhapsodyv, 2020-12-20:
Duplicate of #20387

PR #20545 (2020-12-22): Override STM32 timer assignments to avoid conflicts
### Description

Several STM32 boards had timer conflicts when enabling SPEAKER, BLTOUCH, and TMC2208 drivers with SoftwareSerial.
Override these timer settings in the pins file.
For one board the SoftwareSerial timer number was added in platformio.ini, prevent conflicting with other timers in variant.h.

I have not tested these on hardware, but I did review the PeripheralPins files to check for conflicts with fans.

### Benefits

Fewer complaints about timer conflicts when enabling extra features on boards.

### Configurations

See reference [... cut]
closed 2020-12-20 by rhapsodyv