MarlinFirmware/Marlin #20531: [FR] Faster boot

mrx23dot, 2020-12-21:
### Description
For some reason booting takes ages until main menu is visible on LCD.
6seconds with BOOTSCREENs disabled.
That's a lot for embedded system without OS, which component is taking that long?
USB/serial port is not available during this 6secs, so I don't see it what's going on.

If we don't keep it under control it could grow to 1minute or more in the future.

### Feature Workflow

1. Turn on power supply
2. wait, wait
3. until LCD is ready

#### Additional Information
On SKR 1.3 + ender3. (32-bit ARM Cortex-M3 core running at 96MHz) that's a ton of computing power.

ellensp, 2020-12-21:
the bootloader is looking for a firmware on the sd card for a start

sjasonsmith, 2020-12-21:
It is most likely initializing TMC stepper drivers using Software Serial.

sjasonsmith, 2020-12-21:
To verify this, modify your configuration and comment out your lines defining drivers as TMC2208 or TMC2209 (assuming you have them). I expect it will boot real quick after that.

mrx23dot, 2020-12-21:
Thanks for the feedback.
It's still 6secs without SD card inserted. (surely bootloader exists on first missing SD error)
Do you also see this delay on different platforms?

I only have these enabled, but I was planning to upgrade to TMC, which would add more to this delay.
#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
#define Z_DRIVER_TYPE  DRV8825
#define E0_DRIVER_TYPE DRV8825

qwewer0, 2020-12-21:
Is 6 sec considered slow boot?
My SKR Mini E3 v1.2 boot is around 5+ sec without SD and 5.5-6 sec with SD.

sjasonsmith, 2020-12-21:
I just tried on a Creality 4.2.7 board with a REPRAPDISCOUNT LCD, and it is about 5 seconds as well. This also does not use UART-connected drivers.

I think that in the bigger picture, this is of little consequence. You will spend many hours printing, 5 seconds on boot is negligible. It would be some feat for us to increase this time to one minute, as suggested in the original post.

eightheads, 2020-12-21:
Of all the things to create a ticket for.... wow...

sjasonsmith, 2020-12-21:
Given the low priority of this, I don't think we need to keep a feature request open for it. Any contributor is of course welcome to investigate this and contribute back improvements if this is a priority for them.

closed 2020-12-21 by sjasonsmith

mrx23dot, 2020-12-22:
The extra time happens before setup is called, so that points to the bootloader.

void setup() {
	HAL_init();
	controllerFan.setup(); // pull pin low

During this time the buzzer is constantly on, is that a HW feature?