MarlinFirmware/Marlin #20533: [BUG] FTDI EVE fails to build with HAS_BED_PROBE && !HAS_LEVELING

LinFor, 2020-12-21:
### Bug Description

Marlin fails to build with TOUCH_UI_FTDI_EVE && HAS_BED_PROBE && !HAS_LEVELING

### Configuration Files

Default config with this key modifications:

- TOUCH_UI_FTDI_EVE
- HAS_BED_PROBE

Also need some other edits for run TOUCH_UI_FTDI_EVE, they are not fundamental, for example:

- BOARD_FYSETC_S6
- LCD_FYSETC_TFT81050
- S6_TFT_PINMAP

### Steps to Reproduce

1. Get any configuration, default for example.
2. Set up config options as described in "Configuration Files" previously.
3. Try to build

**Expected behavior:**

Marlin buids.

**Actual behavior:**

Marlin fails to build:
`
In file included from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\../ftdi_eve_lib/extended/ftdi_extended.h:45,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\../ftdi_eve_lib/ftdi_eve_lib.h:27,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\screens.h:25,
                 from Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\advanced_settings_menu.cpp:27:
Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\advanced_settings_menu.cpp: In static member function 'static bool AdvancedSettingsMenu::onTouchEnd(uint8_t)':
Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\advanced_settings_menu.cpp:123:26: error: 'ZOffsetScreen' has not been declared
  123 |     case  2: GOTO_SCREEN(ZOffsetScreen);              break;
      |                          ^~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\../ftdi_eve_lib/extended/screen_types.h:148:51: note: in definition of macro 'GOTO_SCREEN'
  148 | #define GOTO_SCREEN(screen)   current_screen.goTo(screen::onRedraw)
      |                                                   ^~~~~~
*** [.pio\build\FYSETC_S6\src\src\lcd\extui\lib\ftdi_eve_touch_ui\screens\advanced_settings_menu.cpp.o] Error 1
`

thisiskeithb, 2020-12-21:
When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

**This includes attaching your configuration files, regardless of whether you think they are involved.**

LinFor, 2020-12-24:
> When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.
> 
> **This includes attaching your configuration files, regardless of whether you think they are involved.**

I added links to config files.

LinFor, 2021-01-25:
It's actual.

ellensp, 2021-01-25:
Configs need to be the ones that trigger the issue. The linked configs are for a ramps

LinFor, 2021-01-26:
> Configs need to be the ones that trigger the issue. The linked configs are for a ramps

Added configs with key modifications, described in initial post. Also now it's trigger another build fail, in addition to initially described.

closed 2021-03-05 by github-actions[bot]