MarlinFirmware/Marlin #20534: [BUG] Change Filament menu picks wrong Preheat settings on dual extruder machine

idonasch, 2020-12-21:
### Bug Description

The "Change Filament"->"Change/Load/Unload Filament"->"Preheat *" menu selects the wrong preset entry. For E1 always the first (e.g. PLA) preset is used, for E2 always the second (e.g. ABS) is used.
This is related to the closed  issue #20154 that fixed a similar issue with #20201 the "Temperature"->"Preheat *" menu but left the issue in the "Change Filament" menu unaddressed.

### Configuration Files
Customized MBot/Cube configuration for a CTC replicator1 (dual) w/ MightyBoard2560 and 20x4 LCD.
For narrowing down the issue I added a 3rd `PREHEAT_3_*` definition
[CTC.zip](https://github.com/MarlinFirmware/Marlin/files/5725882/CTC.zip)

### Steps to Reproduce
Start with cooled down extruders and distinct  preheat configurations (e.g. 210, 240, 255).
1. Select Load (or Unload) Filament E1, then select 2nd (or 3rd) preset (.e,g. PETG, ABS)
2. check the preset temperature in the main screen
3. the problem is that the temp of the 1st (PLA) preset is used.
4. repeat with E2
5. no matter which preset was selected always the  preset temperature from the 2nd menu is used

**Expected behavior:**

It is expected that the temperature of the selected preset is actually being used.

**Actual behavior:**
a) For E1 always the 1st (PRESET_1_*) settings are used.
b) For E2 always the 2nd (PRESET_2_*) settings are used.
PRESET_3_* can be selected but  see a) or b)
[image]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

sjasonsmith, 2020-12-21:
I thought this had already been fixed. Maybe it was a different menu related to multiple extruders.
@rhapsodyv was it you that fixed something like this earlier?

rhapsodyv, 2020-12-21:
Seems it's a different menu I wasn't aware of. I will take a look.

rhapsodyv, 2020-12-25:
can you test this #20565 ?

idonasch, 2020-12-25:
Works, see also  #20565.

PR #20565 (2020-12-28): Fix Change Filament Menu Actions
### Description

Change filament menu double uses MenuItem::index, leading to one menu level overwriting the other. I tried to fix it on #20201, but I broke the preheat material index.

I'm fixing both cases on this PR.

### Benefits

Fix #20553
Fix #20534
Fix #20154
closed 2020-12-28 by rhapsodyv