MarlinFirmware/Marlin #20535: [FR] (move down until probe triggers)

Phil1988, 2020-12-21:
### Description

I am using a z-min endstop switch and additionally a z probe.
The **z-endstop is connected to the z-min** connector on my SKR 1.3 Board (but the board doesnt matter) and the **probe is connected to the Z-max connector** of the board.
This is wanted for safety reasons, as switches are more robust then Z-probes.
The Z-max connector is mapped as z-probe connection via **#define Z_MIN_PROBE_PIN**.

But it is always **hard to find the perfect z-offset**.
This is due to the reason I can not move until the probe triggers.

At least I did not find a command  (G0-G1 will move until the endstop switches.. not the probe).

I do always need to check M119 mutliple times to get a point where the probe triggers..
The moving and M119 has to performed multiple times and is not as good as it would really trigger

This could be included to something like this:

### Feature Workflow

1. move the probe to a position where it is _not_ triggered.
2. **send a (new) command to move down until the probe triggeres (eg. reprap firmware does it like G1 H1 Z-300 F60)**
3. get the current position with M114
4. manually lower the the z height until to nozzle touches a sheet of paper
5. get the current heigt again with M114
6. do some math et violá.... perfect z-offset :)

I hope someone has an idea of how to do this....
or could include something like this in the future
:)

thisiskeithb, 2020-12-21:
This is already implemented as `PROBE_OFFSET_WIZARD`:

https://github.com/MarlinFirmware/Marlin/blob/7afd274d0ce9c7149c3edf3bd23aadd28154ccfb/Marlin/Configuration_adv.h#L1081-L1095

closed 2020-12-21 by thisiskeithb

Phil1988, 2020-12-22:
But this does only work with Z_SAFE_HOMING which is not wanted. Just a "move down until it triggers and stay there".

Any ideas?

thisiskeithb, 2020-12-22:
> But this does only work with Z_SAFE_HOMING which is not wanted. Just a "move down until it triggers and stay there".

You can disable `Z_SAFE_HOMING` (and the `Z_SAFE_HOMING`/probe sanity check) if you prefer.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

qwewer0, 2020-12-22:
> But this does only work with Z_SAFE_HOMING which is not wanted. Just a "move down until it triggers and stay there".
> 
> Any ideas?

In the latest Bugfix-2.0.x `PROBE_OFFSET_WIZARD` doesn't require `Z_SAFE_HOMING`, just define `PROBE_OFFSET_WIZARD_XY_POS`.

Phil1988, 2020-12-22:
The problem is, that homing (G29) is made in a corner of the print bed where the probe does not trigger.

So should I do this?
1. Heat up the hotend and the heated bed to printing temperature
2. start Z Probe Wizard
3. Wait for homing
4. raise Z and move to a better Position with G1 X100 X100 Z2
5. Move the nozzle down until no light can shine through between the nozzle and the bed
6. Confirm it

If this is the correct way to do it?