MarlinFirmware/Marlin #20536: [FR] Support for syntax highlighters (and compilers) that don't support the extended macro behaviour of gcc

kevquinn, 2020-12-21:
### Description

The Marlin code base contains some use of pre-processor directives inside macro argument expansions, which according to the Microsoft compiler is non-standard. The Microsoft compiler warns about it but accepts it (I think) but the trigger for raising the issue is that it trips up the Visual Studio Code C/C++ code highlighter extension which I imagine is a popular choice of IDE for developers. More generally it may also be desirable to avoid such gcc extensions to be more accommodating of other toolchains, perhaps for less mainstream devices.

Reference [issue I raised with the Visual Studio Code C/C++ code highlighter](https://github.com/microsoft/vscode-cpptools/issues/6685) and the subsequent [issue raised upstream](https://developercommunity.visualstudio.com/content/problem/1293412/c-intellisense-in-gcc-mode-doesnt-correctly-parse.html) on the primary Visual Studio product.

Raising this issue here to gauge appetite to accept pull requests that rewrite occurrences of it in the Marlin code base, and if so to discover if there's a preferred approach to such rewrites before I go ahead working up and submitting some pull requests.

### Feature Workflow

N/A - raising this as a feature issue, as it's not a bug as such but at the same time it's a feature that's only relevant to code developers.

#### Additional Information

Ref. example in [current planner.c](https://github.com/MarlinFirmware/Marlin/blob/f74015b4e57677cd033e116dab2331478edbe2ec/Marlin/src/module/planner.cpp#L1946)

The issue is exemplified in this fragment:

```
      block->millimeters = SQRT(
        #if EITHER(CORE_IS_XY, MARKFORGED_XY)
          sq(steps_dist_mm.head.x) + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.z)
        #elif CORE_IS_XZ
          sq(steps_dist_mm.head.x) + sq(steps_dist_mm.y) + sq(steps_dist_mm.head.z)
        #elif CORE_IS_YZ
          sq(steps_dist_mm.x) + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.head.z)
        #else
          sq(steps_dist_mm.x) + sq(steps_dist_mm.y) + sq(steps_dist_mm.z)
        #endif
      );
```
whereby the `SQRT` macro trips up the Visual Code parser that doesn't support nesting macros (the `#if` clause) inside the expansion of parameters to the SQRT macro.

One way to rewrite this particular case is to move the assignment line and statement closing bracket into the branches of the `#if` clause thus:

```
      #if EITHER(CORE_IS_XY, MARKFORGED_XY)
        block->millimeters = SQRT(sq(steps_dist_mm.head.x) + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.z));
      #elif CORE_IS_XZ
        block->millimeters = SQRT(sq(steps_dist_mm.head.x) + sq(steps_dist_mm.y) + sq(steps_dist_mm.head.z));
      #elif CORE_IS_YZ
        block->millimeters = SQRT(sq(steps_dist_mm.x) + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.head.z));
      #else
        block->millimeters = SQRT(sq(steps_dist_mm.x) + sq(steps_dist_mm.y) + sq(steps_dist_mm.z));
      #endif
```
which Visual Studio Code is happy with.

There are other variations of course; for example:

```
      block->millimeters =
        #if EITHER(CORE_IS_XY, MARKFORGED_XY)
          SQRT(sq(steps_dist_mm.head.x) + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.z));
        #elif CORE_IS_XZ
          SQRT(sq(steps_dist_mm.head.x) + sq(steps_dist_mm.y) + sq(steps_dist_mm.head.z));
        #elif CORE_IS_YZ
          SQRT(sq(steps_dist_mm.x) + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.head.z));
        #else
          SQRT(sq(steps_dist_mm.x) + sq(steps_dist_mm.y) + sq(steps_dist_mm.z));
        #endif
```
although personally I prefer the my first one as I find more readable (of course that's subjective).

kevquinn, 2020-12-31:
Further note; on reading the Marlin coding standards, I see that the it's the standard on [minimising repitition](https://marlinfw.org/docs/development/coding_standards.html#minimize-repetition) being taken to the extreme that leads to the current style in the example I pulled from planner.c.

So I guess I'm not wrong to ask about it before potentially wasting my time on PRs that would likely just be rejected, without prior dicsussion :)

In the example chosen, a minimal change in line with the coding standards - assuming the intent of the standard is not to be so extreme as to abhor the repetition of `SQRT(` - could be:

```
      block->millimeters =
        #if EITHER(CORE_IS_XY, MARKFORGED_XY)
          SQRT(sq(steps_dist_mm.head.x) + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.z))
        #elif CORE_IS_XZ
          SQRT(sq(steps_dist_mm.head.x) + sq(steps_dist_mm.y)      + sq(steps_dist_mm.head.z))
        #elif CORE_IS_YZ
          SQRT(sq(steps_dist_mm.x)      + sq(steps_dist_mm.head.y) + sq(steps_dist_mm.head.z))
        #else
          SQRT(sq(steps_dist_mm.x)      + sq(steps_dist_mm.y)      + sq(steps_dist_mm.z))
        #endif
      ;
```

since if we were to be utterly pedantic, the "+"'s are repetition as much as `SQRT(` and `)` are.

sjasonsmith, 2020-12-31:
Is there a gcc compiler option which would disallow the “offending” macro usage? If so I think you would have an argument for enabling that IF you are willing to fix all the errors it introduces. Without the compiler protecting against it they would likely return again.

You are right to ask before spending too much time on it. There is only one person who could approve such a change, and it is not me...

kevquinn, 2021-01-01:
Did a little more investigation. I had a look at the GNU CPP docs, and there's a pre-processor option [`-Wexpansion-to-defined`](https://gcc.gnu.org/onlinedocs/cpp/Invocation.html#index-Wexpansion-to-defined) that will warn about it (warning enabled also by `-Wpedantic` and `-Wextra`). So I switched it on, and tried a clean build - only to be faced with an almost never-ending stream of warnings...

Turns out it warns on far more than just the occurrences I was worrying about; for example macros MB (_MB_1), PIN_EXISTS (_PINEX_1) trigger it. So I guess that's a no-go; the use of "defined" inside macro expansion is too deeply embedded in the way Marlin is written.

For reference, the rule that's broken according to the [SEI CERT C Coding Standard](https://wiki.sei.cmu.edu/confluence/display/c/PRE32-C.+Do+not+use+preprocessor+directives+in+invocations+of+function-like+macros) is section 6.10.3, paragraph 11 of ISO/IEC 9899:2011. The GNU folks describe it explicitly in their [not bugs list](https://gcc.gnu.org/bugs/#nonbugs_c) as something they support that's non-portable.

Last note for the moment however; comments on the [upstream Visual Studio bug report](https://developercommunity.visualstudio.com/content/problem/1293412/c-intellisense-in-gcc-mode-doesnt-correctly-parse.html) suggest it's something intellisense can handle in later versions - so it may well be something they'll resolve.

In any case, I'll close this issue off with no change, as it doesn't seem viable and we're better off hoping that Visual Studio Code gets an update to intellisense.

closed 2021-01-01 by kevquinn

sjasonsmith, 2021-01-01:
Thanks for looking into that and providing the update. I know from past projects that increasing the warning verbosity of any code base tends to be non-trivial.

In my own projects I tend to favor enabling all warnings and treating them as failures, but that is much easier on a new codebase.