MarlinFirmware/Marlin #20541: [BUG] Dual Extruder Setup on SKR 1.3 -- Unable to move E0 at all. E1 works properly.

RFBomb, 2020-12-22:
### Bug Description

- I cannot move E0 at all if I have 2 nozzles configured. 
- E1 works as expected (Movement occurs when using G1 E after usinig the T1 command to select E1). 
- E0 shows no movement, even after using the T0 command to select it. All other actions work properly (temperature reading and setting, as well as XY moving into position after the T command.)
- If I revert back to 1 nozzle and comment out the necessary items (E-Steps, PID, driver, etc), then E0 works properly. 

### Configuration Files

[Dual_Nozzle_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5727842/Dual_Nozzle_Configuration.zip)

### Steps to Reproduce

1. Attempt to use 2 nozzles on a single X carriage in config
2. Attach to printer with octoprint / pronterface for better debugging
3. Learn that E0 doesn't work. 

**Expected behavior:**

E0 moves as expected after selecting it with T0 command. 

**Actual behavior:**

E0 stepper does not drive. Temperature setting works properly. 

#### Troubleshooting already performed

- Motors check out OK. Tested by connecting cable from E1 motor to E0 motor. E0 motor is driven perfectly fine using the E1 driver. 
- Cables check out OK. Tested by connecting the E0 cable into the E1 driver. This can drive either E motor just fine. 
- Stepper Driver tested OK. Tested 3 different stepper drivers. All exhibited same symptoms.
- Board tested OK. I disabled nozzle 2 (E1) in the config, recompiled and loaded it so that the printer was back to a single-nozzle setup. This was then using the E0 driver. This worked as expected, with no issues driving the stepper motor. 
 
[Single_Nozzle_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5727841/Single_Nozzle_Configuration.zip)

#### Additional Information

TMC Debug info from when 2 nozzles were configured:

```
SENDING:M122 I S0
		X	Y	Z	E	E1
Address		0	0	0	0	0
Enabled		true	true	true	true	true
Set current	500	500	800	800	1000
RMS current	489	489	795	795	994
MAX current	689	689	1121	1121	1402
Run current	15/31	15/31	25/31	25/31	17/31
Hold current	7/31	7/31	12/31	12/31	8/31
CS actual	7/31	7/31	12/31	25/31	8/31
PWM scale
vsense		1=.18	1=.18	1=.18	1=.18	0=.325
stealthChop	true	true	true	true	true
msteps		16	16	16	16	16
tstep		max	max	max	max	max
PWM thresh.	57	56	19	60	81
[mm/s]		175	176	103	13177	100
OT prewarn	false	false	false	false	false
triggered
 OTP		false	false	false	false	false
pwm scale sum	24	23	40	53	26
pwm scale auto	2	2	2	1	0
pwm offset auto	88	90	92	64	94
pwm grad auto	31	27	34	14	14
off time	4	4	4	4	4
blank time	24	24	24	24	24
hysteresis
 -end		2	2	2	2	2
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
uStep count	120	760	168	200	40
DRVSTATUS	X	Y	Z	E	E1
sg_result	142	0	64	0	0
stst
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:07:00:00
		Y	0xC0:07:00:00
		Z	0xC0:0C:00:00
		E	0x40:19:00:00
		E1	0xC0:08:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing E connection... OK
Testing E1 connection... OK
```

ellensp, 2020-12-22:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

RFBomb, 2020-12-22:
I’ll test that tomorrow If I have time. I took a quick look at whatever
commit notes there were and didn’t see anything seemingly related to my
problem though.

ellensp, 2020-12-22:
I cannot reproduce.
I used marlin 2.0.7.2 and your config files on a genuine SKR 1.3 with BTT TMC2209's in UART mode.
The only thing I changed was that I disabled your CR10_STOCKDISPLAY and enabled REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
My test was simply enable cold extrusion M302 P1 (since these are just stepper motors, not hotend or extruders) 
and click the extrude and reverse button on pronterface. 
E0 moved as expected
Extrude sends 
G91
G1 E5.0 F100
G90
reverse sends
G91
G1 E-5.0 F100
G90
I then send a T1 and clicked extrude and retract again
this time E1 moved as expected.

ellensp, 2020-12-22:
My best guess is you have the wrong connectors on your stepper motors (ie plugs that do not have the two clips to hold it in place) and you managing to consistently have a bad connection on E0

RFBomb, 2020-12-22:
I’ll try again tonight, thank you for testing. As far as the connections
go, The cables are good, and it has the same connectors as the stock
cables. I even checked the crimps, and they all look good.I also did the
wiggle test once it was moving and it never stopped.

 Plus, without changing any wiring (Or touching it even), I modified the
config to the 1-nozzle version Started moving E0 just fine. So I’m a bit
baffled as to why it wasn’t moving E0 during my testing yesterday.

Since you were able to get it working, I’ll try out my current firmware
again and update to bug fix if problem continues and report back.

RFBomb, 2020-12-23:
I have recompiled firmware using 2.0.7.2 and the configuration for Dual-Extruder that I uploaded in the OP, and restored all the wiring to how I plan to run it (same state as it was originally set up when issue was seen). 
Everything is now working as expected. It is possible that I had a bad connection to E0, and swapping the plugs around resolved it. Unsure what the actual problem was, but that appears to be the most likely culprit (despite the connections, crimps, etc all looking correct...)

Thank you for testing on your end. I'm closing this issue now as the root problem has been resolved.

closed 2020-12-23 by RFBomb