MarlinFirmware/Marlin #20542: custom user menu

legend069, 2020-12-22:
### Configuration Files
 
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5728216/configs.zip)

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

after uploading firmware and running the cusom g code- preheat. once preheated i have to abort the print?

**Expected behavior:**

select custom menu to preheat then go back into cusom menu to adjust z offset
**Actual behavior:**

select custom menu to preheat then i have to abort the print that isn't real then i can adjust z offset
#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

legend069, 2020-12-22:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5728235/config.zip)

uploaded the cusom user menu. seperatly

ellensp, 2020-12-22:
You have
` #define USER_GCODE_1 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND) "\nM117 E/B HEATING STARTED \nM300 S440 P200 \nM300 S440 P200"
`
which has no terminating \n

you also have weird spaces

Try this. 
`#define USER_GCODE_1 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND) "\nM117 E/B HEATING STARTED\nM300 S440 P200\nM300 S440 P200\n"
`

ellensp, 2020-12-22:
nope never mind... that is no longer correct...  shouldn't have terminating \n  wish they would stop changing things...

legend069, 2020-12-22:
> 
> 
> Ignore previous reply. that is no longer correct... shouldn't have terminating \n wish they would stop changing things...

noted. what determins the funcitons for the "abort print " to appear - the funtion that tells the pinter that it has a print ?
( hoping you understand my terrible explaination. )

i'm thinking it would be something like "if gcode has more than 3 lines" print job status true

sjasonsmith, 2020-12-22:
Please provide exact steps to reproduce this. By exact I mean exactly what you are selecting on the screen at each step.

legend069, 2020-12-22:
> 
> 
> Please provide exact steps to reproduce this. By exact I mean exactly what you are selecting on the screen at each step.

1: press button
2: scroll to "Custom Menu"
3: press button
4: scroll to "Heat ALL beeep ABS"
5: press button
6:wait for it to beep twice
7: attempt to adjust z offset but the "abort print option shows up

sjasonsmith, 2020-12-22:
Your Configuration.h is version 020008 and Configuration_adv.h is version 020007.

I cannot compile them with current bugfix-2.0.x, due to the stale Configuration_adv.h.

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

If the problem still exists in `bugfix-2.0.x` we will need updated configuration files which compile against the bugfix branch.

legend069, 2020-12-23:
i can confirm using the version 020008 has not fixed the error. i have made a small change to the custom menu and i belive i have found it to be the "M109" and "M190" command if i send them seperatly using octoprint terminal the "abort print" shows up.

new error with new firmware; the printer no longer beeps after completing the actions in custom menu.
i've re downloaded bugfix-2.0.x. copied over my config.h used the example for my printer/board and made the changes to config_adv.h 020008  copied changes from 020007

the after disabling "//#define SPEAKER" in config.h it beeps again but not the frequencies.
i can confrim i was able to produce tones using the previous firmware.

sjasonsmith, 2020-12-24:
@legend069 can you provide the configuration files you used with the bugfix-2.0.x branch?

legend069, 2020-12-24:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5737886/Configuration.zip)

uploaded

LinFor, 2020-12-24:
I also confirm this behavior, and it seems it has always been.
Using M109 / M190 (blocking, with waiting) in custom commands gives this behavior (the printer thinks that printing has started), while M104 / M140 (non-blocking) does not. Perhaps it has something to do with the print timer.

My configs, if you need it: https://github.com/LinFor/Marlin/blob/e-bot-s6/Marlin/Configuration.h

rhapsodyv, 2020-12-28:
It is the expected behaviour. Marlin doesn't know what is a printing. It just execute g-code. But, some g-code are linked to a printer job timer. For example: M75 will start this timer.

If you have `PRINTJOB_TIMER_AUTOSTART` enabled, some g-code normally used on printing will start the timer too. It is very well explained in the Configuration.h:

```cpp
/**
 * Print Job Timer
 *
 * Automatically start and stop the print job timer on M104/M109/M190.
 *
 *   M104 (hotend, no wait) - high temp = none,        low temp = stop timer
 *   M109 (hotend, wait)    - high temp = start timer, low temp = stop timer
 *   M190 (bed, wait)       - high temp = start timer, low temp = none
 *
 * The timer can also be controlled with the following commands:
 *
 *   M75 - Start the print job timer
 *   M76 - Pause the print job timer
 *   M77 - Stop the print job timer
 */
#define PRINTJOB_TIMER_AUTOSTART
```

closed 2020-12-28 by rhapsodyv