MarlinFirmware/Marlin #20543: [BUG] 4 EXTRUDERS and dual servo SWITCHING_EXTRUDER

DrumClock, 2020-12-22:
Hi,
this error occurs when setting` #define EXTRUDERS 4 `and two servos for `#define SWITCHING_EXTRUDER` for E0/1 and E2/3.

[error.txt](https://github.com/MarlinFirmware/Marlin/files/5728430/error.txt)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5728426/Configuration.zip)

Tested on bugfix 2.0.x from 22.12.2020

ellensp, 2020-12-22:
no. Invalid incomplete bug report. Once again trying to slip a feature request for an unsupported hardware configuration in as a bug.

closed 2020-12-22 by ellensp

DrumClock, 2020-12-22:
Hi @ellensp 
The error will be in the module ... \ Marlin \ src \ inc \ Conditionals_LCD.h in the section

 ```
 #if DISABLED (SWITCHING_NOZZLE)
     #define HOTENDS E_STEPPERS
   #endif

```
Here, HOTENDS is set to 2 even if there are physically 4 extruders.
If I disable `#define HOTENDS E_STEPPERS`, everything is fine.

sjasonsmith, 2020-12-22:
@DrumClock I think you are right. Can you submit a pull request with that change?

I (and probably most of us) don't even know how to configure a switching extruder. It would be best for the fix to come from someone actually using it.

DrumClock, 2020-12-22:
HI @sjasonsmith 
Here is my whole branch: https://github.com/DrumClock/Marlin/tree/GTR-4EX2
I don't know exactly what you need, so look at that.

**Unfortunately, the module does not work after today's update**

... `Marlin \ src \ HAL \ STM32 \ timers.cpp`: 105: 31: error: 'NUM_HARDWARE_TIMERS' was not declared in this scope
   105 | HardwareTimer * timer_instance [NUM_HARDWARE_TIMERS] = {nullptr};

sjasonsmith, 2020-12-22:
Here is an introduction to pull requests:
https://marlinfw.org/docs/development/getting_started_pull_requests.html

You already described the code issue a few comments up. You are apparently the only person using the combination of features you have, so it is important for you to learn how to submit back fixes when you find them.

As for that timer issue, that is only happening on your end. Keith and I have both already built your configurations and it isn't a problem.

sjasonsmith, 2020-12-22:
I thought you were correct, but now I agree with @ellensp that this is more of a feature request.

In it's current form I think `SWITCHING_EXTRUDER` implies that each pair of filaments use a single HOTEND. Changing this would break any existing users of this feature, unless there are exactly zero successful `SWITCHING_EXTRUDER` users today.