MarlinFirmware/Marlin #20546: [BUG] Thermistor Table 66 changed since 1.1.9

woodencase01, 2020-12-22:
### Bug Description

Dyze Design 500°C thermistor has a pretty high resistance value at room temperature.
It used to work well with version 1.1.9, but not anymore with 2+.
Most of the time, the heaters are inactive, and a mintemp error is triggered.

### Configuration Files

Marlin-2.0.7.2\Marlin\Configuration_adv.h
```
MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10
MILLISECONDS_PREHEAT_TIME 30000
```

### Steps to Reproduce

1. Set a thermistor to 66
2. Send a heat command

**Expected behavior:**

Should start heating, for at least 30 secs.

**Actual behavior:**

Min temp error

#### Additional Information

Marlin\thermistortable_66.h has changed from 2+ version. The two last lines from the table used to be:
```
  { OV(1023) - 1, 25},
  { OV(  1023),  20}
```
and now (Marlin 2+) are:
```
  { OV(  1022),  25 },
  { OV(  1023),  20 }
```

ellensp, 2020-12-22:
Makes almost zero difference. infact  { OV(  1022),  25 }, produces a smoother curve  
-1 looks to have been in error see spreadsheet of final table after all macros have been applied 
https://docs.google.com/spreadsheets/d/1jSzT2OCkWQRHxW8wC07YgHOtcKbXwOlBIOOLyYc47Y0/edit?usp=sharing

woodencase01, 2020-12-22:
> Makes almost zero difference. infact { OV( 1022), 25 }, produces a smoother curve
> -1 looks to have been in error see spreadsheet of final table after all macros have been applied
> https://docs.google.com/spreadsheets/d/1jSzT2OCkWQRHxW8wC07YgHOtcKbXwOlBIOOLyYc47Y0/edit?usp=sharing

I have to disagree.

The high resistance at room temperature return both values of 1023 and 1022 from the ADC, but more often 1023. Considering the oversampling total, the result gets more often closer to 20°C.

To keep the min_temp safety, we set it to 21°C. However, if the table sets the 25°C at 1022 oversampled, we get min_temp error too often.

By using { OV( 1023)-1, 25 }, we ensure that if there is one read at 1022 during the sampling, the temperature will be considered 25°C and thus, not triggering a false min_temp error.

Keep in mind that this sensor is using the limit of the ADC, where this tweak was tested and working, while still keeping the machine safe. Having an open circuit will always read 1023 with the pull-up.

sjasonsmith, 2020-12-22:
Please attach your configuration files, as required by the issue template.

sjasonsmith, 2020-12-22:
Possibly related to #20546.

I have one of these thermistors sitting here (well, a clone at least), but will wait for the configuration files to be posted to try anything with it.

ellensp, 2020-12-23:
converted to 10bit adc values 

```
temp | current   | old
---------------------------
30   | 1021.25   | 1021.25
25   | 1022.9375 | 1022
20   | 1023      | 1023
```

If this is rounding up.. this may be an issue

RNewDeal, 2020-12-30:
Interestingly enough I use the T-D500 from Trianglelabs and the thermistor table, along with following the steps outlined by dyze and the listing, have it function perfectly fine. Could it possible be the tables being changed to suit the TL thermistor? Or a possibly overlooked setting in marlin?

sjasonsmith, 2020-12-31:
It looks like this was changed in the following PR, which added thermistor 67:
https://github.com/MarlinFirmware/Marlin/pull/13275

There is no mention of this thermistor's change. I'm guessing it was assumed that `(OV(1023) - 1) == OV(1022)`, which is not true.

That change may or may not have been intentional, but I'm going to close this as not a bug anyway.

My reason is that this tweak you mentioned works you you, but isn't a universal solution. Your room-temperature is such that it works. My room temperature is 18C, so it wouldn't work for me.

These thermistors simply cannot work reliably at room temperature, which is why the options you've listed in the description exist. We should not add hacks to the tables to allow thermistors to work depending on your room's thermostat setting.

You are of course more than welcome to modify the table yourself for your own usage.

closed 2020-12-31 by sjasonsmith