MarlinFirmware/Marlin #20547: Do not compile with Arduino Mega and Windows

Gustavo-Forster, 2020-12-22:
Marlin 2.0.7.2 do not compile with Windows 10 running Arduino 1.8.13 board MEGA 2560 
Geting error:
avr-gcc: error: CreateProcess: No such file or directory
exit status 1
Erro compilando para a placa Arduino Mega or Mega 2560

pinchies, 2020-12-22:
This error is generic, and it sounds like an error with Arduino IDE, not with Marlin. Can you compile other arduino sketches successfully?

thisiskeithb, 2020-12-22:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-22 by thisiskeithb

Gustavo-Forster, 2020-12-22:
Hi.
Thanks for the prompt answer.
Yes I can compile Marlin 2.6.0
Seems to be a problem with Windows

Enviado do Email<https://go.microsoft.com/fwlink/?LinkId=550986> para Windows 10

ellensp, 2020-12-22:
This is a Arduino IDE on Windows issue. Windows has a hard coded unchangeable max path/fork limit of 32K characters. Arduino IDE is written badly, when it attempts to link all the .o files together to form the final .bin file on large projects, such as Marlin, it generate one very long command that can easily exceed the windows limit.

There are a number of tricks to shorten the path under windows. 
Move marlin to root of drive, rename it to m
Delete unused parts of marlin such as unused HAL directories and unused display types. But this require in depth knowledge of how marlin works to know what is safe to remove.

Ultimately it is far better to use PlatformIO on vscode instead of Arduino IDE
PlatformIO only complies and links the code you need for your system, this avoids the Windows limitation. It also produces smaller binaries and is faster.

Gustavo-Forster, 2020-12-22:
Thanks, I'll teu Platformio

Obter o Outlook para Android<https://aka.ms/ghei36>