MarlinFirmware/Marlin #20553: All Load/Unload Filament Menu Items Use First Preheat Setting

thisiskeithb, 2020-12-23:
### Bug Description

All "Load Filament" & "Unload Filament" options under the "Change Filament" menu use `PREHEAT_1_TEMP_HOTEND` instead of their defined `PREHEAT_*_TEMP_HOTEND` temperatures.

### Configuration Files

Any config with the steps listed below will work, but here's one of mine rebased to the latest `bugfix-2.0.x` (28440867721b5c614db604c6c6e82c4dd7ab41cb): https://github.com/thisiskeithb/Marlin/tree/myconfig/MK3S-BTT002-TMC2209

### Steps to Reproduce

1. Enable multiple preheat constants like:
```cpp
// Preheat Constants
#define PREHEAT_1_LABEL       "PLA"
#define PREHEAT_1_TEMP_HOTEND 210
#define PREHEAT_1_TEMP_BED     65
#define PREHEAT_1_FAN_SPEED   255 // Value from 0 to 255

#define PREHEAT_2_LABEL       "PETG"
#define PREHEAT_2_TEMP_HOTEND 245
#define PREHEAT_2_TEMP_BED     85
#define PREHEAT_2_FAN_SPEED   255 // Value from 0 to 255

#define PREHEAT_3_LABEL       "CF PC"
#define PREHEAT_3_TEMP_HOTEND 290
#define PREHEAT_3_TEMP_BED    105
#define PREHEAT_3_FAN_SPEED   255 // Value from 0 to 255
```
2. Enable `M600`/Advanced Pause enabled as well as `FILAMENT_LOAD_UNLOAD_GCODES` so the menu item "Change Filament" is enabled.
3. Go to "Change Filament" -> "Load Filament" or "Unload Filament" and then select a preheat other than the first one.
4. Every "Load Filament" option will use `PREHEAT_1_TEMP_HOTEND` instead of the correct `PREHEAT_*_TEMP_HOTEND` setting.

**Expected behavior:**

All "Load Filament" or "Unload Filament" options to use the correct `PREHEAT_*_TEMP_HOTEND` setting.

**Actual behavior:**

All "Load Filament" or "Unload Filament"  options incorrectly use `PREHEAT_1_TEMP_HOTEND` setting.

#### Additional Information

Preheats under the "Temperature" menu work correctly.

LinFor, 2020-12-24:
Dublicate of #20534 ?

thisiskeithb, 2020-12-24:
> Duplicate of #20534 ?

Related, but this is on a single extruder builds.

rhapsodyv, 2020-12-25:
can you test this #20565 ?

thisiskeithb, 2020-12-25:
> can you test this #20565 ?

That fixed it.

closed 2020-12-25 by thisiskeithb

sjasonsmith, 2020-12-25:
@thisiskeithb you don’t want to wait for a fix to actually merge before closing this?

PR #20565 (2020-12-28): Fix Change Filament Menu Actions
### Description

Change filament menu double uses MenuItem::index, leading to one menu level overwriting the other. I tried to fix it on #20201, but I broke the preheat material index.

I'm fixing both cases on this PR.

### Benefits

Fix #20553
Fix #20534
Fix #20154