MarlinFirmware/Marlin #20554: (FR)  4 EXTRUDERS and dual servo SWITCHING_EXTRUDER

DrumClock, 2020-12-23:
I thought you were correct, but now I agree with @ellensp that this is more of a feature request.

In it's current form I think `SWITCHING_EXTRUDER` implies that each pair of filaments use a single HOTEND. Changing this would break any existing users of this feature, unless there are exactly zero successful `SWITCHING_EXTRUDER` users today.

_Originally posted by @sjasonsmith in https://github.com/MarlinFirmware/Marlin/issues/20543#issuecomment-749752076_

DrumClock, 2020-12-23:
HI @sjasonsmith
I tried to adjust the FW according to the pictures below.
I'm not a programmer, but on the first try it seems to work.
I tried different configurations of SWITCHING_EXTRUDER, SWITCHING_NOZZLE, HOTENDS and everything could be compiled.

Can someone please check it in tests?

[image]

[image]

sjasonsmith, 2020-12-23:
You really need to keep the templates when you post issues and feature requests.

DrumClock, 2020-12-23:
**Current behavior:**
`SWITCHING_EXTRUDER` only supports 1 HOTEND
for 2 HOTENDS  must also be enabled `SWITCHING_NOZZLE`.

**Behavior after adjustment:**
`#define HOTENDS 4 // Physical Hotends for all switching extruder`
a `#if DISABLED (SWITCHING_NOZZLE) &&! HOTENDS `can define how many physical HOTENDS you really need independently
SWITCHING_NOZZLE function. This is useful, for example, when the servo switches EXTRUDER and NOZZLE at the same time and 2 HOTENDS are used.

DrumClock, 2021-01-03:
Hi @sjasonsmith 

the **extruder and nozzle switching function with the same servo** is tested and functional.
The option to switch the extruder with one hotend remains after deactivation ` #define HOTENDS 4` retained.

Please edit the FW (I can't create a PR).

DrumClock, 2021-01-08:
> Hi @sjasonsmith
> 
> the **extruder and nozzle switching function with the same servo** is tested and functional.
> The option to switch the extruder with one hotend remains after deactivation ` #define WITH_NOZZLE` retained.
> 
> Unfortunately, I don't know how to edit the `TOOLCHANGE_ZRAISE` function, which should end after the `SWITCHING_EXTRUDER` function.

**to the section in Configuration_adv.h**

```
/ **
  * Universal tool change settings.
  * Applies to all types of extruders except where explicitly noted.
  * /
#if HAS_MULTI_EXTRUDER
   // Z raise distance for tool-change, as needed for some extruders
   #define TOOLCHANGE_ZRAISE 2 // (mm)
   // # define TOOLCHANGE_ZRAISE_BEFORE_RETRACT // Apply raise before swap retraction (if enabled)
   // # define TOOLCHANGE_NO_RETURN // Never return to previous position on tool-change
   #if ENABLED (TOOLCHANGE_NO_RETURN)
     // # define EVENT_GCODE_AFTER_TOOLCHANGE "G12X" // Extra G-code to run after tool-change
   #endif

```
**add option**
`#define ZRAISE_WITH_SWITCHING_EXTRUDER`

Enabling this function would return the Z axis until the tools were changed and the extruder was switched.

closed 2021-01-30 by DrumClock