MarlinFirmware/Marlin #20555: [FR] Add custom LCD abort print script

GameFan01, 2020-12-23:
I'd like you to make it possible to customize what the printer will do if you stop the print from the LCD menu. For example to present the print by moving Y to the max position or unload filament.

Sorry if this is not the right place to post this, as this is the first time I'm writing here.
Thank you for your time.

thisiskeithb, 2020-12-23:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/20513

thisiskeithb, 2020-12-23:
You can already customize Marlin's actions when you abort a print. Just change `G28XY` to whatever you want:

https://github.com/MarlinFirmware/Marlin/blob/f74015b4e57677cd033e116dab2331478edbe2ec/Marlin/Configuration_adv.h#L1184

closed 2020-12-23 by thisiskeithb

GameFan01, 2020-12-24:
Thanks for the info.