MarlinFirmware/Marlin #20559: [BUG] SD card - file listing - wrong modification time

mrx23dot, 2020-12-24:
### Bug Description
SD card file listing is not based on true modification time. 
(looks like it behaves more like creation time)

### Configuration Files
set:
#define SDCARD_RATHERRECENTFIRST
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/5739554/conf.zip)

### Steps to Reproduce
1. Have min 3 STL files on SD card
2. Modify the oldest one

**Expected behavior:**
The modified oldest one should be on top.

**Actual behavior:**
The file remains at the previous order (not top).

The SD card shows the correct file on top when in insert it into PC.
I started a different print because of this multiple times, and new users will also fail for this.

rhapsodyv, 2020-12-24:
There's no bug here. The docs says:
```cpp
  // Reverse SD sort to show "more recent" files first, according to the card's FAT.
  // Since the FAT gets out of order with usage, SDCARD_SORT_ALPHA is recommended.
  #define SDCARD_RATHERRECENTFIRST
```

So, `SDCARD_RATHERRECENTFIRST` is not related to mtime or ctime, but in the FAT table index.

closed 2020-12-24 by rhapsodyv

mrx23dot, 2020-12-24:
Ahh your comment is clearer than the one in the code.