MarlinFirmware/Marlin #20560: [BUG] Print starts without finishing ABL sometimes

newbie238, 2020-12-24:
https://user-images.githubusercontent.com/76560827/103091314-d7d7f500-4604-11eb-9627-48da96a98399.mp4

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/5739705/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/5739706/Configuration.txt)

Roughly 10% of every try, my print starts without finishing auto bed leveling. And if its going to do that, it only happens after 4th probing point. 90% prints i do not encounter any problem. I am not sure this is called glitch or a bug.

FanDjango, 2020-12-24:
Fix mounted probe, ok.
Suggestion: Compile with
```
/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
#define DEBUG_LEVELING_FEATURE
```
and turn this on by putting `M111 S32` into your print start GCODE before the G29 ABL, and maybe turn it off with `M111 S0` after the G29. And then wait for the failure. In the debug output one might see if and why the probing is failing.

thisiskeithb, 2020-12-24:
It's likely silently failing to probe & then continues on, but:

Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

rhapsodyv, 2020-12-25:
Try disable `#define EXTRAPOLATE_BEYOND_GRID` and test again.

closed 2021-02-02 by github-actions[bot]