MarlinFirmware/Marlin #20563: [FR] CHOPPER_09STEP_12V

kad, 2020-12-24:
### Description

It would be good to have CHOPPER_09STEP_12V similar to CHOPPER_09STEP_24V.
I'm looked for some documentation on how to define properly CHOPPER_TIMINGs for 0.9 degree motors running on 12V printer, but without success. I see how it is defined for many other variants in Marlin/src/features/tmc_util.h, but don't see any pointers on how to calculate or determine those...  Any pointer to right direction? Or maybe someone will be able to provide PR with description on how to calculate those?

### Feature Workflow

1. Printer with 12V PSU and TMC drivers.
2. Change motors from usual 1.8 degree/step to 0.9 degree/step.
3. #define CHOPPER_TIMING ....
4. there is CHOPPER_09STEP_24V that can be used, but there is no CHOPPER_09STEP_12V and there are no links to docs on how to calculate or determine those.

#### Additional Information

Would be happy to contribute PR for those defines, if someone points on documentation how to calculate those.

ellensp, 2020-12-25:
This sets the data in the structure and used when initialising the stepper driver.
typedef struct {
  uint8_t toff;
  int8_t hend;
  uint8_t hstrt;
} chopper_timing_t;

currently we have
#define CHOPPER_DEFAULT_12V  { 3, -1, 1 }
#define CHOPPER_DEFAULT_19V  { 4,  1, 1 }
#define CHOPPER_DEFAULT_24V  { 4,  2, 1 }
#define CHOPPER_DEFAULT_36V  { 5,  2, 4 }
#define CHOPPER_PRUSAMK3_24V { 3, -2, 6 }
#define CHOPPER_MARLIN_119   { 5,  2, 3 }
#define CHOPPER_09STEP_24V   { 3, -1, 5 }

These are registers in the tcm stepper driver and can be found in the datasheet eg https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC2224_datasheet_Rev1.09.pdf
See section 5.4.1 CHOPCONF

If you look in the tmcstepper repo you find 
https://github.com/teemuatlut/TMCStepper/blob/master/examples/Calibrate_spreadCycle/Calibrate_spreadCycle.ino

run this on a test system running 12v and 0.9 steppers

LinFor, 2020-12-25:
> I see how it is defined for many other variants in Marlin/src/features/tmc_util.h, but don't see any pointers on how to calculate or determine those... Any pointer to right direction? Or maybe someone will be able to provide PR with description on how to calculate those?

Every TMC datasheet has detailed step-by-step guide how to determine this parameters and fine-tune for your motor. For example, in this (https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC220x_TMC2224_datasheet_Rev1.09.pdf) datasheet see pages 62-64 (chapter 14).

kad, 2020-12-31:
@LinFor  thanks for pointing right place in datasheet.
@ellensp Thanks. Unfortunately I don't have test system that is "arduino compatible". I tried to make that sketch to properly working on BTT SKR mini E3, but so far no luck :(

Anyone tried to incorporate that "spreadcycle calibration" into Marlin 2.0.x HAL ?

freighttt, 2021-01-09:
i too would be interested in a way to run the calibration program

moswald14, 2023-08-28:
Did you ever find a good stealthchop settings for your 12v powered 0.9degree steppers? I asked chatGPT for some configs to start out and I will be testing them:

```
Certainly, here are a few examples of different combinations of StealthChop settings in the format { <off_time>, <hysteresis_end>, <hysteresis_start> } that you can try as starting points for your 0.9-degree stepper motors running on 12V. Remember that these are just starting points, and you might need to fine-tune the values based on your specific setup and requirements:

Example 1 - Conservative Settings:
off_time: 10
hysteresis_end: 2
hysteresis_start: 3

Example 2 - Balanced Settings:
off_time: 9
hysteresis_end: 4
hysteresis_start: 4

Example 3 - Smooth Operation:
off_time: 12
hysteresis_end: 5
hysteresis_start: 2

Example 4 - Low Noise at Low Speeds:
off_time: 8
hysteresis_end: 3
hysteresis_start: 5

Example 5 - High-Speed Performance:
off_time: 6
hysteresis_end: 8
hysteresis_start: 1

Example 6 - Customized Balance:
off_time: 7
hysteresis_end: 6
hysteresis_start: 3

Remember to start with lower currents and gradually increase them while monitoring motor temperature and behavior. Additionally, observe the noise level and smoothness of operation at different speeds. It's important to find a balance between noise reduction and torque/speed performance that suits your application.

Make sure to consult the documentation of your specific motor driver and controller for guidance on how to adjust these settings, as the process may differ depending on the hardware you're using.
```

moswald14, 2023-08-29:
Update: I tried a couple from the list GPT3.5 gave me and they didnt seem to go anywhere, so I told it its settings didn't work and it asked me for details about my specific setup and hardware, I put in the specifications and the numbers is spit out for me actually work. they're not completely silent, but they are to the point where its not going to give me a headache.

> User
> Im running a corexy with 0.9 degree steppers at 32x microstepping and 12v 1.8amps with TMC2209 drivers on belted x and y axis with 20 tooth pulleys. I want balanced performance for my 3d printer
> specs for the stepper:
> * Manufacturer Part Number: 17HM19-2004S
> * Motor Type: Bipolar Stepper
> * Step Angle: 0.9°
> * Holding Torque: 46Ncm(65.1oz.in)
> * Rated Current/phase: 2.0A
> * Phase Resistance: 1.45ohms
> * Recommended Voltage: 12-24V
> * Inductance : 4.0mH±20%(1KHz)

```
ChatGPT
Thank you for providing more details about your setup. Given your desire for balanced performance for your 3D printer using the provided stepper motor (17HM19-2004S) and TMC2209 drivers, I can suggest StealthChop settings to help achieve smoother and quieter operation while maintaining good performance. Here's a recommended set of values to start with:

Microstepping: 32x
Motor Current: 1.8A (You can start with the rated current of the motor, but you might need to adjust it based on temperature and performance.)
StealthChop Settings:
```

```
off_time: 8
hysteresis_end: 4
hysteresis_start: 3
```

thisiskeithb, 2026-09-28:
Give https://github.com/MarlinFirmware/Marlin/pull/28617 a try.

closed 2026-10-20 by thinkyhead