MarlinFirmware/Marlin #20567: [BUG] Sensorless homing still does homing bump despite having homing bump for X and Y axis set to 0

NovaViper, 2020-12-25:
### Bug Description

When I run G28, my printer normally would move backoff the X and Y axis first by 5mm for the sensorless homing and then go ahead and home each axis and then once it hits the frame of the printer it stops. However with the recent v2.0.7.2 release, I noticed that it's no longer doing it, but instead seems to try to home printer *before* performing the sensorless homing backoff. The behavior it ends up doing now is quite similar to how the homing bump works, but I have that setting turned off because I have TMC2209s with sensorless homing enabled. And because of this, if my printer is already at its homing position and I go and try to home, it grinds against the frame. This happens on both the release branch and the bugfix branch.

### Configuration Files
[MarlinBugfix.zip](https://github.com/MarlinFirmware/Marlin/files/5741148/MarlinBugfix.zip)
[MarlinStable.zip](https://github.com/MarlinFirmware/Marlin/files/5741187/MarlinStable.zip)

### Steps to Reproduce

1. Enable sensorless homing and releavent settings
2. Set homing bump to 0 and enable sensorless backoff
3. Flash printer and home
4. Notice how the printer does not perform the backoff, but instead performs a homing bump

**Expected behavior:**

Perform sensorless homing backoff first and then attempt homing

**Actual behavior:**

Homes first before doing sensorless homing backoff, resulting in a normal homing bump.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

Compressed video of behavior
[20201224_225114.zip](https://github.com/MarlinFirmware/Marlin/files/5741164/20201224_225114.zip)

Speaka, 2020-12-25:
You're probably looking for `SENSORLESS_BACKOFF_MM`? It is set to 5 in you config. Or is it real bump, with two homing moves and triggered endstops?

NovaViper, 2020-12-26:
> You're probably looking for `SENSORLESS_BACKOFF_MM`? It is set to 5 in you config. Or is it real bump, with two homing moves and triggered endstops?

That's what it was before I updated the firmware, for some reason it does the bumping despite I have the homing bump turned off entirely (it's set to 0 for the homing bump setting). For sensorless homing it's recommended to have the homing bump turned off as it can throw off the firmware, and to avoid the firmware crashing the bed/hotend into the frame it's recommended to have the firmware back off each axis first (which is what that sensorless_backoff setting does); which for some reason the latest update completely ignores both settings

NovaViper, 2020-12-26:
Did some testing today, the printer doesn't do any homing back when I have sensorless_backoff_mm turned off.

NovaViper, 2020-12-28:
Hmm.. even odder is the abnormal behavior only occurs when I run `G28`, which runs homes all of the axes. But it doesn't occur when I home each axis individually

rhapsodyv, 2020-12-28:
Try disabling `QUICK_HOME`

NovaViper, 2020-12-28:
Yep! The odd behavior goes away when I disable `QUICK_HOME`. Something changed with that code in a recent commit (however I can't find which one messed it up specifically because I accidentally overwritten my old configuration 😢)

rhapsodyv, 2020-12-28:
I remember I saw it something, sometime ago... that `QUICK_HOME` don't play well with backoff and bump. But I didn't look further.

NovaViper, 2020-12-28:
Hmm.. It was working flawlessly with the TMC2209s for quite a while until very recently when I updated my local branch to the very latest bugfix commit. Wonder what it was that changed it 🤔

thisiskeithb, 2020-12-29:
> I remember I saw it something, sometime ago... that `QUICK_HOME` don't play well with backoff and bump. But I didn't look further.

I tried to create a sanity check for it at one point, but it wasn’t merged: https://github.com/MarlinFirmware/Marlin/pull/17084.

I’ve run into this issue on sensorless homing builds (as well as other quirks), so I leave `QUICK_HOME` disabled for those and recommend others do the same when they’re having homing issues.

closed 2021-02-06 by github-actions[bot]