MarlinFirmware/Marlin #20569: [avr-gcc: error: Creating process] (When Marlin is compiled, both in VS and Arduino IDE, throws this error.)

AayushBoliya, 2020-12-25:
### Bug Description

### Configuration Files

[Config. Files Drive link](https://drive.google.com/drive/folders/1wM02yUiJTfk5nzM0n_h4nOULItcnTPDZ?usp=sharing)

Parametric changes of stepper motors and hotend maxtemp.

### Steps to Reproduce

1. [Install fresh Martin zip file from official website or github repository.]
2. [Open Marlin.ino in Marlin folder.]
3. [Compile/Verify]

**Expected behavior:**

**Actual behavior:**

sjasonsmith, 2020-12-25:
This is a limitation of the maximum command line length when using Arduino on Windows. There are ways to work around it, but it is usually a better option to switch to Visual Studio Code with PlatformIO.

This has many advantages over Arduino;
- Avoid command length issues
- MUCH faster builds
- Automatically dependency management
- Benefits from all future improvements to Marlin build processes

[Installing Marlin (VSCode)](https://marlinfw.org/docs/basics/install_platformio_vscode.html)
[Auto Build Marlin](https://marlinfw.org/docs/basics/auto_build_marlin.html)

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-12-25 by sjasonsmith

Roxy-3D, 2020-12-26:
I wonder if we should change the documentation and just say Arduino is no longer supported as a build environment?

Alternatively...  It does kind of seem to me the Arduino group would be interested in using Marlin as a build environment and test each version against it.   Marlin is one of the more complicated Arduino projects and it would provide them with a good test for new releases.