MarlinFirmware/Marlin #20577: Intermittent Auto Home Failed and Manual push Axis X or Y stop trigger Printer Halted with A10M GEEETECH

mmaitre74, 2020-12-26:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Upgrate to last firmware Marlin 2.0 Bug Fix
2. In the LCD menu select Auto Home (intermittent Home Failed and Printer Halted)
3. For Manual forcing problem while pressing LCD Auto HOME, click any Endstop Switch (X,Y)

**Expected behavior:**

If endstop switch pressed, the printer must stop axis pressed and continu to going Home point to other axis

.

**Actual behavior:**

The Marlin version 1.1.8 and 1.1.9 has no problem with the ends
[configa10m.zip](https://github.com/MarlinFirmware/Marlin/files/5743650/configa10m.zip)
top or intermittent problem

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

godbnc, 2020-12-27:
i have same problem with cr10v2 not home error i check configuration ,endstop,wires conection, but not there is problem  with marlin .....

ellensp, 2020-12-27:
homing is not that simple (depending on setting) ...  if you press and hold an endstop it will fail as it thinks the endstop is broken.
It need to be able to home, back off and home again getting a triggered, not triggered, and re triggered in sequence.
I'm not saying this is the issue, but I am saying your test of holding an endstop in a triggred state is not necessarily a good test.

mmaitre74, 2020-12-27:
I found a solution. Everything work fine now.

I remove this line in configuration.h

// Validate that endstops are triggered on homing moves
// #define VALIDATE_HOMING_ENDSTOPS

rhapsodyv, 2020-12-27:
can you try enabling `#define ENDSTOPPULLDOWNS` instead of `#define ENDSTOPPULLUPS`?

mmaitre74, 2020-12-28:
#define ENDSTOPPULLDOWNS , same issue

mmaitre74, 2020-12-28:
Version 1.18 and 1.19 woking without issue.

https://user-images.githubusercontent.com/19422624/103245622-bccff280-492e-11eb-8c73-364b42f66ac5.mov

FanDjango, 2020-12-28:
Have you got a 1.1.9 config to show?

FanDjango, 2020-12-28:
And in the video I see that there is a small "stop" in the middle of the homing move of X

FanDjango, 2020-12-28:
Check playing with this setting, maybe:
```
/**
 * Endstop Noise Threshold
 *
 * Enable if your probe or endstops falsely trigger due to noise.
 *
 * - Higher values may affect repeatability or accuracy of some bed probes.
 * - To fix noise install a 100nF ceramic capacitor in parallel with the switch.
 * - This feature is not required for common micro-switches mounted on PCBs
 *   based on the Makerbot design, which already have the 100nF capacitor.
 *
 * :[2,3,4,5,6,7]
 */
//#define ENDSTOP_NOISE_THRESHOLD 2
```

FanDjango, 2020-12-28:
I see the homing bump of X, also of Y. Hmmm.  Do me a favor: Do the homing one by one, first Home X, then Home Y, then Home Z. Isolate the problem

mmaitre74, 2020-12-28:
Hi FanDjango, thanks for your help. with #define ENDSTOP_NOISE_THRESHOLD 2. No more issue. Everything work perflectly.

closed 2020-12-28 by mmaitre74