MarlinFirmware/Marlin #20580: [FR] - EXTRA_FAN_SPEED - logic change

MasterPIC, 2020-12-27:
M106 S0 ; fan off
M106 T255 ; fan still off
M106 T2 ; switch to alternate speed mode - fan at max as expected
M106 S128 ; fan at 50% - **I'd prefer fan to stay at max**
M106 T1 ; fan **off** - **I'd prefer fan to switch from max to 50%**

In a few words I'd prefer
M106 T1 activating "normal" speed;
M106 T2 activating "extra" speed;
M106 Sn changing normal speed value;
M107 resetting both normal and extra speed values.

swilkens, 2020-12-27:
Changing what "S" does is not practical, many slicers / slicer profiles out there use it to set the fan speeds.

sjasonsmith, 2020-12-27:
@MasterPIC can you provide an example of your workflow when you would need this? I wasn't even aware this `T` parameter existed until today, and I'd like an example of when you would choose to use it.

I would also like to know whether you have a slicer that actually generates these, or if you are inserting them in your own custom scripts.

MasterPIC, 2020-12-27:
I may be wrong but I think that no slicers currently use extra fan speed except by postprocessing script.
I considered extra fan speed as a feature useful to manipulate gcode produced by a slicer.
For postprocessing: the user can manipolate the gcode to enter/exit from the extra_speed mode without taking care of the M106 S commands produced by the slicer...

MasterPIC, 2020-12-27:
@sjasonsmith I can for example prefer to have fan at 30/255 on perimeter. Well, this feature is not commonly available.
I can simply set M106 T30 and replace (by postprocessing) "outer perimeter" by M106 T2.
Then there is the need for switching back to usual fan mode by M106 T1. What can be done by replacing the comment used to indicate a different part of extrusion by an M106 T1...
I'm obviously referring to verbose Gcode.

MasterPIC, 2020-12-27:
@sjasonsmith Another example: sometimes You can realize that for the ongoing print it's a good idea to keep fan off but printing already started. Well, while printing You can send just two commands to tell the printer to keep the fan off...

sjasonsmith, 2020-12-27:
So the real problem is that your post-processor has added `M106 T2`, but any fan changes from the slicer itself end up throwing you back into "normal fan mode", as if you had called `M106 T1`?

sjasonsmith, 2020-12-27:
Looking at the code, I think there are lots of edge cases with strange behavior regarding this `EXTRA_FAN` functionality.
Unfortunately I can't find any examples using Google of people actually using this, to see how changing behavior might impact people. Maybe you are the only one using it!

For example:
```
M106 S255; Fan at 100%
M106 T1; Restores speed that was never saved, so fan turns off.
```

```
M106 S255; Fan at 100%
M106 T2; Switch to extra speed that was never set, fan turns off. This might be expected.
```

```
M106 S255; Full Speed
M106 T128; Set alternative 50% speed
M106 T2; Enable alternate speed, fans go to 50%
M106 S0; Fan turns off
M106 T1; Disable alternate speed, fan turns back to 100%

sjasonsmith, 2020-12-27:
If behavior were changed so that `M106 S` impacts only "normal" fan speed and `M106 T` impacts only "extra" fan speed, it is less obvious what `M107` should do.

If this goal is for `M106 T2` to cause all normal fan commands from the slicer to be ignored, then it seems like you would also want to ignore `M107`.

MasterPIC, 2020-12-28:
I see that current implementation makes it useless.

When I found this new feature available I assumed it worked as I'm still expecting.
I think it can be useful if the printer keeps track of all M106 Sn commands so that the user can simply insert (at line k of the preprocessed gcode)

M106 Tm (with m>2)
M106 T2 (to fix speed at m)

and at line k+n(+2) the code
M106 T1 (to revert to the slicer fan speed)

to manually set an user managed fan speed from line k to line k+n without affecting the following (slicer managed) part of the gcode...

So I see it as a possible feature to override slicer fan speed. Current implementation makes it causing a mix of speeds without any usefulness...

Tupson444, 2023-05-07:
I've turned EXTRA_FAN_SPEED in the firmware (2.1.2) and I use "G-code substitutions" in PrusaSlicer (just in case I need to adjust fan speed during print, and to avoid it resetting every time after bridging which is set to higher fan speed).

I want to share my thoughts about this, as it was confusing and difficult to set it correctly.

First, the explanation for setting secondary speed is confusing (M106 T3-255) – I thought: _alright, so "restore fan speed" is T1, "use secondary speed" is T2, then "set secondary speed" is T3 and then add value from 0 to 255_, therefore I set the GCODE literally "M106 T3-255" and I was confused why it doesn't work. It would be better to have a separate letter for setting the value, not like this which limits the minimum to 3 (so you can never set 0% fan speed). I know it's not important as it does work this way, but it's a bit clumsy IMO.

Second, I had to use tricks in sequence of GCODE commands in order to set it working as intended. I thought that I can first set the primary and secondary fan speed, then use M106 T1 and M106 T2 for future switching between the two. However, when I set say M106 S178, and then use M106 T1, I doesn't set to 178 but 0. I was again confused what is going on, until I realized that it sets to _previous_ fan speed. It might be better to change it so there is a way to set "primary fan speed" and then call it whenever you like afterwards, not just switching to whatever was previous fan speed.

In the end, this is how I set the slicer (PrusaSlicer 2.5.0) to work properly:

- Under Filament Settings > Cooling i set "fan speed" to 70%, "bridges fan speed" to 100%, "disable fan for first" 1 layer and "full fan speed at layer" 3.
- Under Print Settings > Output options > G-code substitutions I added the following:

> find:
M106 S89.25
replace with:
M106 T255\nM106 S178\nM106 T2\nM106 T1

>find:
M106 S178.5
replace with:
M106 T1

>find:
M106 S255
replace with:
M106 T2

As it is set to start fan at layer 2 at half speed (35% = M106 S89.25) and full speed at layer 3 (70% = M106 S178.5), at layer two, first the secondary fan speed is set (M106 T255), then the fan is set to 70% (M106 S178), then it's switched to secondary fan speed (M106 T2), then it is swapped back to "previous fan speed" (M106 T1). Only this way I managed to make it switch between 70% for normal printing and 100% for bridging.

If it was coded differently (being able to set "primary fan speed", not "previous fan speed"), it would be easier to set up slicer by setting primary and secondary fan speed in the start code, and then use them later.