MarlinFirmware/Marlin #20582: [BUG] FTDI EVE fails to build with AUTO_BED_LEVELING_BILINEAR

Lino77, 2020-12-28:
BOARD_FYSETC_S6_V2_0
LCD_FYSETC_TFT81050
S6_TFT_PINMAP

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5747102/Marlin.zip)

Linking .pio/build/FYSETC_S6/firmware.elf
/home/lino/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/FYSETC_S6/src/src/lcd/extui/lib/ftdi_eve_touch_ui/marlin_events.cpp.o: in function `ExtUI::onMeshUpdate(signed char, signed char, float)':
/home/lino/Drucker/fysetc_s6/Firmware/Marlin_2.0.8/Marlin/src/lcd/extui/lib/ftdi_eve_touch_ui/marlin_events.cpp:144: undefined reference to `BedMeshScreen::onMeshUpdate(signed char, signed char, float)'
/home/lino/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/FYSETC_S6/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/screens.cpp.o:(.rodata._ZN9ScreenRef13functionTableE+0x340): undefined reference to `BedMeshScreen::onEntry()'
/home/lino/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/FYSETC_S6/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/screens.cpp.o:(.rodata._ZN9ScreenRef13functionTableE+0x350): undefined reference to `BedMeshScreen::onRedraw(draw_mode_t)'
/home/lino/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/FYSETC_S6/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/screens.cpp.o:(.rodata._ZN9ScreenRef13functionTableE+0x354): undefined reference to `BedMeshScreen::onTouchStart(unsigned char)'
/home/lino/.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/FYSETC_S6/src/src/lcd/extui/lib/ftdi_eve_touch_ui/screens/screens.cpp.o:(.rodata._ZN9ScreenRef13functionTableE+0x35c): undefined reference to `BedMeshScreen::onTouchEnd(unsigned char)'
collect2: error: ld returned 1 exit status
*** [.pio/build/FYSETC_S6/firmware.elf] Error 1
================================================================================== [FAILED] Took 42.47 seconds ==================================================================================

Environment    Status    Duration
-------------  --------  ------------
FYSETC_S6      FAILED    00:00:42.472
============================================================================= 1 failed, 0 succeeded in 00:00:42.472 =============================================================================
The terminal process "pio 'run', '--environment', 'FYSETC_S6'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

sjasonsmith, 2021-01-01:
@Lino77 can you try the change from the above PR?

This does disable the mesh edit screen entirely for your build using Bilinear. An early change restricted it to only be enabled with UBL, but the change wasn't quite complete.

sjasonsmith, 2021-01-04:
A fix for this has been merged. I'm closing this, but please reply if you still have issues compiling.

closed 2021-01-04 by sjasonsmith

Lino77, 2021-01-06:
it works thanks