MarlinFirmware/Marlin #20584: [BUG] MAX31865 library call not working for GTR V1.0 board with PIN values above 127

GadgetAngel, 2020-12-28:
### Bug Description

For the GTR V1.0 board PIN values above 127 will not work with the Adafruit MAX31865 library call because the Adafruit MAX31865 library uses 'int8_t' to pass the PIN number where as the GTR V1.0 framework uses 'uint32_t' for the PIN number this causes an OVERFLOW to occur for PINs who value is above 127.  The following PINS are effected:

PH12,
PH13,
PH14,
PH15, 
PI0,
PI1
PI2,
PI4,
PI5,
PI6, and
PI7

### Configuration Files

[Configuration GTR V1.0 and MAX31865 lib BUG.zip](https://github.com/MarlinFirmware/Marlin/files/5747817/Configuration.GTR.V1.0.and.MAX31865.lib.BUG.zip)

### Steps to Reproduce

1. Use a GTR V1.0 Board
2. See wiring Diagram
3. in configuration.h set `#define TEMP_SENSOR_0 -5`
4. in pins_BTT_GTR_V1_0.h file:
```
#define TEMP_0_PIN          PH13
#ifndef MAX6675_SS_PIN
   #define MAX6675_DO_PIN                    PI2
   #define MAX6675_SCK_PIN                   PI1
   #define MAX31865_MOSI_PIN                PI10
  #define MAX31865_CS_PIN                    TEMP_0_PIN
  #define UNUSED2_PIN                            PI9
  #define MAX6675_SS_PIN                       UNUSED2_PIN   //must be different for MAX31865_CS_PIN to get software SPI call
#endif
```
This ends up with **MIN TEMP** ERROR due to the fact that MAX31865_CS_PIN, MAX6675_DO_PIN and MAX6675_SCK_PIN are OVERFLOW when they go into the MAX31865 library call.  MAX31865_CS_PIN value is 129, MAX6675_DO_PIN is 134 and MAX6675_SCK_PIN is 133.  All these value are over 127 so they get wrap around to negative values which in turn gets read as a very large numbers when GTR V1.0 framework uses in the `digitalWrite` routine for the framework:
```
void digitalWrite(uint32_t ulPin, uint32_t ulVal)
{
  digitalWriteFast(digitalPinToPinName(ulPin), ulVal);
}
```

The call to `digitalPinToPinName()` returns a `NC` because the negative number is read as a `uint32_t` value and it can not find that value in the array `digitalPin[p]`.  Same applies for the call to `pinMode()` in the framework and `digitalRead()` in the framework.

Wiring Diagram:

[image]

**Expected behavior:**

I expected my MAX31865 to work as it does on the SKR PRO V1.1 board.

**Actual behavior:**

I get an immediate **MIN TEMP** ERROR.

rhapsodyv, 2020-12-28:
It's not a marlin bug. Neither a stm32duino bug. But a MAX31865 library issue. The library could not assume anything about pins. 
We can't do nothing on Marlin... and I doubt stm32duino will change anything...

GadgetAngel, 2020-12-28:
> It's not a marlin bug. Neither a stm32duino bug. But a MAX31865 library issue. The library could not assume anything about pins.
> We can't do nothing on Marlin... and I doubt stm32duino will change anything...

The LPC framework handles it just fine.  Two pin on the LPC framework are above 127 and the LPC framework handle it. In the LPC framework instead of using 'uin32_t' for indexing the PIN values, the LPC framework just uses 'int16_t' for the pin values and then just looks at the bottom byte and ignores the rest of the upper 8 bits. The LPC framework masks out the pin value to only look at the first 8 bits because most board pin values are between 0 and 255.

So that puts the Marlin user of the GTR V1.0 board in a pickle. 

 Marlin is not willing to fix this issue with a conversion layer?

sjasonsmith, 2020-12-29:
I'll re-open this for further consideration, but can't promise when anything might happen with it, or that it will not be closed again.

GadgetAngel, 2021-01-28:
this issue has not yet been fixed

GadgetAngel, 2021-02-10:
This issue is resolved because the PR #20447 has now been merged in bugfix branch

closed 2021-02-10 by GadgetAngel