MarlinFirmware/Marlin #20585: Disabling SHOW_BOOTSCREEN causes TFT_COLOR_UI builds to grow in size

thisiskeithb, 2020-12-28:
### Bug Description

Disabling `SHOW_BOOTSCREEN` with `TFT_COLOR_UI` causes builds to grow and potentially fail to build on smaller boards.

This bug is two-fold because:

1. `BOOT_MARLIN_LOGO_SMALL` is no longer checked (and thus "disabled") in this block when `SHOW_BOOTSCREEN` is disabled:
https://github.com/MarlinFirmware/Marlin/blob/a8c361c93bdda3110d147c5f47f96f57298c0a13/Marlin/Configuration_adv.h#L1141-L1146
2. `TFT_COLOR_UI` _always_ compiles in a bootscreen, which means the larger of the two bootscreens due to item 1 above.

While trying to fix the issue, I added a check around the TERN (in `ui_480x320.h` for example) which works:
```cpp
  #if ENABLED(SHOW_BOOTSCREEN)
    TERN(BOOT_MARLIN_LOGO_SMALL, MarlinLogo195x59x16, MarlinLogo480x320x16),
  #endif
```
But `tft_image.h` isn't happy when I try to do something similar to fix the index:
```cpp
  TERN(SHOW_BOOTSCREEN, imgBootScreen = 0x00, imgHotEnd = 0x00),
  imgBed,
  imgBedHeated,
```

I'm still looking into this and have recruited @rhapsodyv to help 🙂

### Configuration Files

[Robin Nano 1.2 - TFT_COLOR_UI.zip](https://github.com/MarlinFirmware/Marlin/files/5748027/Robin.Nano.1.2.-.TFT_COLOR_UI.zip)

### Steps to Reproduce

1. Use the attached configs or build for a board with a small flash size & `TFT_COLOR_UI` & `SHOW_BOOTSCREEN` disabled.
2. See build fail due to too much flash space being used.

**Expected behavior:**

Build to succeed with `SHOW_BOOTSCREEN` disabled and not be larger than with it enabled.

**Actual behavior:**

Build fails with `SHOW_BOOTSCREEN` disabled due to `TFT_COLOR_UI` always compiling in a bootscreen and now using the larger of the two.

#### Additional Information

Originally reported by @th0m4sek: https://github.com/MarlinFirmware/Marlin/commit/cfcfc8047afb09bd3da8d3e7bb49f066a977e6d6#commitcomment-45465331.

filimonic, 2020-12-28:
Thank you!
Confirming: Firmware is NOT building successfully in case of 
```
TFT_COLOR_UI = ON
SHOW_BOOTSCREEN = OFF
```

```
#define TFT_COLOR_UI
//#define SHOW_BOOTSCREEN => 
ld.exe: region `rom' overflowed by 76740 bytes

#define TFT_COLOR_UI
#define SHOW_BOOTSCREEN =>
RAM:   [========  ]  78.6% (used 51504 bytes from 65536 bytes)
Flash: [=====     ]  54.2% (used 284188 bytes from 524288 bytes)

```

closed 2020-12-29 by thisiskeithb

thisiskeithb, 2020-12-29:
This has been fixed, so pull the latest `bugfix-2.0.x`.

New results using an ET5X config with a TFT35/TFT24 and `TFT_COLOR_UI`:

## TFT35
### No bootscreen
```prolog
Flash: [==        ]  21.6% (used 226668 bytes from 1048576 bytes)
```
### Small bootscreen
```prolog
Flash: [==        ]  23.8% (used 249772 bytes from 1048576 bytes)
```
### Large bootscreen
```prolog
Flash: [=====     ]  50.9% (used 533956 bytes from 1048576 bytes)
```

## TFT24
### No Bootscreen
```prolog
Flash: [==        ]  21.2% (used 222236 bytes from 1048576 bytes)
```
### Small Bootscreen:
```prolog
Flash: [==        ]  23.4% (used 245356 bytes from 1048576 bytes)
```
### Large Bootscreen:
```prolog
Flash: [====      ]  35.9% (used 375940 bytes from 1048576 bytes)
```