MarlinFirmware/Marlin #20587: [BUG] Bilinear leveling not compensating

GraydonLeed, 2020-12-28:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Run abl ( G29)
2. store settings to overwrite any EEPROM meshes 
3. Start a print

**Expected behavior:**

The bed will be level and abl will compensate for warping

**Actual behavior:**

Does 0 compensating and will be uneven

Tried new firmware, I'm on the latest version, cleared EEPROM, reconfigured abl with more points, 

[Marlin-2.0.7 bugreport (3).zip](https://github.com/MarlinFirmware/Marlin/files/5748358/Marlin-2.0.7.bugreport.3.zip)

thisiskeithb, 2020-12-28:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

sjasonsmith, 2020-12-28:
I have spent many hours testing and disproving the claim that ABL "Does 0 compensating". At this point it is hard for me to believe these claims, and the problem is most likely in your process.

Please provide _exact_ steps to reproduce. Every single step. Starting with turning on the power switch. Also include a gcode file you are trying to print as well.

GraydonLeed, 2020-12-28:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
>
> * Download Marlin [...]

 Tried doing this and during probing the Z-axis just went all the way up. even after homing 3 times.

FanDjango, 2020-12-28:
Z Axis going all the way up sounds very much like: Z-Axis direction needs to be reversed, quite honestly. Need more detail.
Does the same thing happen when you home Z?

GraydonLeed, 2020-12-28:
> Z Axis going all the way up sounds very much like: Z-Axis direction needs to be reversed, quite honestly. Need more detail.
> Does the same thing happen when you home Z?

Nope. Z homes properly

FanDjango, 2020-12-28:
Ok. Z homes ok. Gimme the configuration.h only in here to let me look at it.

GraydonLeed, 2020-12-28:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5749126/Configuration.zip)
I homed it a few more times and it probed 3 points, then just went back up all the way again

FanDjango, 2020-12-28:
What means "all the way"? To the physical limit (krack-krack-krack)?

GraydonLeed, 2020-12-28:
Yep. Physical limit, not firmware limit

FanDjango, 2020-12-28:
Ok, so you have `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`, and homing works.
So Z stepper direction looks ok.
It is probing that fails on G29. Can you try M48 instead?

FanDjango, 2020-12-28:
Config 020007 tells me it is not bugfix. I hate to ask, but maybe one should go to bugfix to be sure

GraydonLeed, 2020-12-28:
Damn i thought i had it on bugfix. My bad. Will recompile

FanDjango, 2020-12-28:
Still not sure what the problem is to make the Z axis go up all the way suddenly, if ALL ELSE (like homing) works. I would really like to isolate the problem, by homing XY and Z, and then doing M48, which ALSO does probing, but with no other movement (in the default).

GraydonLeed, 2020-12-28:
Recv: echo:Unknown command: "M48"

FanDjango, 2020-12-28:
My config version on bugfix is `#define CONFIGURATION_H_VERSION 020008` for confirm
Ah, M48 unknown. Need to enable. 
```
// Enable the M48 repeatability test to test probe accuracy
//#define Z_MIN_PROBE_REPEATABILITY_TEST
```

sjasonsmith, 2020-12-29:
If you still have issues after testing with `bugfix-2.0.x` please upload new configuration files also. Just `Configuration.h` and `Configuration_adv.h`.

FanDjango, 2020-12-29:
@sjasonsmith take over, I am going horizontal now, it's after 1 am...

GraydonLeed, 2020-12-29:
Got the bugfix branch working, testing abl now

GraydonLeed, 2020-12-29:
Bed leveling still does not work. Will put the probe output here.
[g29 output.txt](https://github.com/MarlinFirmware/Marlin/files/5749329/g29.output.txt)

rhapsodyv, 2020-12-29:
Did you set the Z Offset before print or G29?

GraydonLeed, 2020-12-29:
z offset was before g29

Oninaig, 2020-12-30:
@GraydonLeed so it completes the creation of the bilinear mesh and its ENABLED when printing, but its not actually compensating or rather, not compensating enough? I am having a similar issue with bilinear where I can probe the bed just fine and the mesh is valid, but when I go to print its like it doesn't use the mesh values properly resulting in a first layer that is way too close on the right side of the plate and too far everywhere else.

GraydonLeed, 2020-12-30:
> @GraydonLeed so it completes the creation of the bilinear mesh and its ENABLED when printing, but its not actually compensating or rather, not compensating enough? I am having a similar issue with [...]

Exactly, i do seem to have it be working a little.

sjasonsmith, 2020-12-31:
The probed bed looks pretty good to me. It certainly needs to be manually trammed better, but it looks "real". It is a mostly-flat surface with a clear tilt. This does not look like random data.

I noticed that you do not have `EXTRAPOLATE_BEYOND_GRID` enabled. Without this feature, the edges the bed cannot reach will not continue the slope of the bed. This will cause your prints in those regions to be too low or high on two edges of your bed. This would be very noticeable with your bed, since you have quite a bit of slope to it.

GraydonLeed, 2020-12-31:
Will enable that, Thank you!

thestephenmarshall, 2021-01-13:
@GraydonLeed did that work? If you run the "Mesh Validation Sequence" as seen [here](https://marlinfw.org/docs/gcode/G026.html), do you get the same results? You will need `#define G26_MESH_VALIDATION` un-commented for this to work. Here's an example where my nozzle is `260C`, my bed is `100C` and my desired layer height is `0.2` : `G26 B100 H260 L0.2`

rhapsodyv, 2021-01-13:
@GraydonLeed you can enable M114_DETAIL, then move the head to a position, and call `M114 D`, it will show what is applied.

GraydonLeed, 2021-01-13:
Hi guys, i have moved to UBL because i can make easier manual adjustments for that.

closed 2021-01-13 by GraydonLeed