MarlinFirmware/Marlin #20590: [BUG] strange behavior on z-axis

Cysign, 2020-12-28:
### Before Reporting a Bug

1. `bugfix-2.0.x` branch: I couldn't compile bugfdix-branch...as I had other problems with the RA control panel and an i2c driven HD44780 I decided to solve those problems not using the buxfixbranch.
Is there a list of all changes the bugfixes made?
Or maybe this is a known bug?

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at 
Here's what I've done so far with cura files, gcode files, marlin config-files, and a video showing the behavior and the display.

https://reprap.org/forum/read.php?415,878907,879958#msg-879958
and
https://community.ultimaker.com/topic/35343-z-offset-after-auto-levelingrafting-z-level-relativeabsolut/?tab=comments#comment-279736

Asumption is it's a faulty ocnfig on my printer, as some guy on the cure board could print my testfile with proper bottom layers.

### Bug Description
Hi,
I finally finished my 3D printer on the hardware. But configuring still makes some trouble.
I use automated-bad-leveling using an inductive probe.
On each print I first home my axis, then do a bed leveling and then the print starts properly with rafting (as I set up using Cura or Slic3r).
During the whole rafting-process z-level on the display is 4.65mm.
When the object starts printing, my z-level still is 4.65mm for the next layers (so I had to set the filament flow to 70% to avoid the extruder's gear grinding the filament too much...but I'll handle this as soon as z-level works properly).
At this height the raft shuts the nozzle, while there's no filament flow (although I can see extruder movement as well as grinded filament).

All layers with z-level smaller than 4.65mm are lost. I can see the x/y-movements on the print, but as the nozzle is locked by the raft, there's no filamentflow.
Once the z-level is greater than 4.65mm everything works like exspected:

Picture of what's going wrong:
https://content.invisioncic.com/ultimake/monthly_2020_12/77.thumb.jpg.14c4c43736061c36d0515960e1949949.jpg
(top let: I marked what's missing on my prints, top right:  that's what all my prints look like: the first 4.65mm are missing...but then it's printing well, bottom: that's what my prints look like - though I didn't set the shonw -4.65 on z-axis on my prints. I just made this picture to understand the behavior).

Here are my files: 
http://www.digitalinventions.de/files/temp/IronBot_Cura_Marlin_Problem_first_4.65mm.zip

Here's the video (showing display as well as printinghead movements...sorry for tha bad quality): 
https://youtu.be/vQkE3Z16EZI

### Steps to Reproduce

As the ultimaker community link shows, some other guys can't reproduce the problem using my files.
So it's either a configuration-bug on my machine (especially in the auto-leveling process) or it's a Marlin bug....but I couldn't confirm either :(

zeleps, 2021-01-04:
I have a similar problem on my machine, a custom parking extruder configuration with a positive z-offset on the 2nd extruder. When a 2nd-extruder-only print starts, Z indication is 0.75 (the second extruder's z-offset), and the layers below that height are printed at actual height 0. When printing layer @0.9, height begins to raise normally. Gcode is verified to provide the correct z position for all layers. Since the dual extruder setup is new and I don't have much experience with it, I'm not sure it the z value on info screen always displays an absolute value (based on the first extruder's height) when printing with the second extruder or it should be the second extruder's translated z-value.

I'll have more tests conducted tomorrow, including a dual print and some tests without ABL and I'll also disable NO_WORKSPACE_OFFSETS mode (which is active now in my config) to see if the problem persists, and I'll come back with more info.

Cysign, 2021-01-06:
Still no progress :( Now I'm trying to deactivate several marlin features to find out where the problem occurs.

Cysign, 2021-01-22:
Looks like the bugfix-branch has solved my problem.
Thanks for your work, guys ;)

closed 2021-01-22 by Cysign