MarlinFirmware/Marlin #20594: [BUG] inconsistent space beetween M117 and M118 hostactions

ndelpl, 2020-12-29:
### Bug Description

When sending a M117 notification command with host command enabled, its convert into
//action:notification text
 into host terminal
When sending the same via M118  command, its convert into
// action:notification text
The space beetween // and action is inconsistent and need complex parsing in host.
In my case, i try to create a custom menu into the TFT35 lcd for acting with octoprint.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Send M117 text 

Send M118 P0 E1 text

Send M118 P0 A1 text

And see the space diferencies

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**
The same space :either with or without, but co sistent

**Actual behavior:**
One space in one case, no space in the other case

#### Additional Information

I don t push merge request for now because i don't know if the right code is with or without space.

Thank for reading.

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2020-12-29:
that majority seem to use SERIAL_ECHOPGM("//action:"); part of host_action  
So to me it simply looks like `if (hasA) SERIAL_ECHOPGM("// ");` in M118.cpp has an extra space.

ndelpl, 2020-12-29:
Tanks for answer,
Im agree with that! ( i made the same modification on my code repo.)

rhapsodyv, 2020-12-29:
Ok, it must be fixed.

But is a bit strong to say that one space at the beginning of the line leads to complex parsing... 😉

ndelpl, 2020-12-31:
> Ok, it must be fixed.
> 
> But is a bit strong to say that one space at the beginning of the line leads to complex parsing... 😉

Im agree its a bit strong, the parser involve is the parser from TFT35 that is not as powerfull ( its a microcontroleur too) as the linux/octoprint ones.
Anyway, thanks for taking acount of this request

ellensp, 2021-01-02:
merged and closed

closed 2021-01-02 by ellensp