MarlinFirmware/Marlin #20595: [BUG] With PROBING_HEATERS_OFF G28 and G29 do not wait for the heaters to reach temperature after completion

Sebazzz, 2020-12-29:
### Bug Description

When having `PROBING_HEATERS_OFF` enabled for the new Creality CR-6, G28 and G29 do not wait for the heaters to come back on temperature before returning. This can cause prints using these commands in the start gcode to extrude material at a too low temperature.

### Configuration Files

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5751194/Configs.zip)

### Steps to Reproduce

1. CR-6 SE printer
2. Set #PROBING_HEATERS_OFF
3. Run a print with G28 (and optional G29) in the start gcode and a heat to 235C *before* the G28

**Expected behavior:**

After G28 and the temperatures are dropped, the print immediately begins without waiting for the temperatures to stabilize. Especially with PETG and ABS prints this is a problem.

**Actual behavior:**

Before G28 and G29 - first wait on the heaters. Preferably only do this when we can expect additional commands.

#### Additional Information

I've patched this in these commits:
- https://github.com/CR6Community/Marlin/commit/598c9f837e5697c7349bba9b6203f7b995fd6ab3
- https://github.com/CR6Community/Marlin/commit/d4e7e739234a2a26cdb5bd785a205597233fea9e

But this approach is not ideal. The idea is that if we are printing or have other commands in the queue, we first wait for the hot-end to reach temperature. If no other commands are pending or we're not printing, we can just continue and skip the heating to save some time.

Thoughts?

slowbro, 2021-01-21:
A possible solution for this in the context of the `probe` module is to add a config to wait for the hotend, too. That would mean that between probes, it would be keeping the hotend at temperature, and therefore there would be much less of a thermal deficit when probing is finished. This way, the problem is solved for all things that use the probe and the code to ensure temperature won't have to be added everywhere.

For the time being, you can always add a `M109` after your `G28`/`G29`. I can make a PR for the `WAIT_FOR_HOTEND` or similar setting.

Sebazzz, 2021-01-28:
That would significantly increase probing time because Marlin waits quite a while before temperatures are stabilized.

slowbro, 2021-01-28:
> That would significantly increase probing time because Marlin waits quite a while before temperatures are stabilized.

Indeed, I noticed this when testing it. It's been a busy week at the dayjob, but I'm still thinking about a better solution on the backburner.

closed 2021-03-07 by github-actions[bot]