MarlinFirmware/Marlin #20597: [BUG] Z_STEPPER_AUTO_ALIGN doesn't work anymore

BastR, 2020-12-29:
### Bug Description

Z_STEPPER_AUTO_ALIGN doesn't work anymore, if I emit a G34, printer makes homing XYZ then deploy the Bltouch to start the G34 procedure but I've got "probing failed"

### Configuration Files

https://github.com/BastR/Marlin/tree/V-Core-Pro

Send: G34 _(printer make G28 XYZ before)_
[...]
Recv: X:245.00 Y:200.00 Z:5.17 E:0.00 Count A:35600 B:3600 Z:2068
Recv: 
Recv: G34 Iteration: 1 _(bed is moving down, Bltouch deploy)_
[...]
Recv: Probing failed
Recv: G34 aborted. _(then printer make G28 Z)_
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
[...]
Recv: X:245.00 Y:200.00 Z:5.17 E:0.00 Count A:35600 B:3600 Z:2068
Recv: ok

BastR, 2020-12-29:
@sjasonsmith

BastR, 2020-12-29:
Resetting the eprom has solucioned the problem....

Sorry

closed 2020-12-29 by BastR

sjasonsmith, 2020-12-29:
There is an open issue related to automatically selected points being stored to the EEPROM. I prototyped something to avoid that, but have not yet gone back to finish it.