MarlinFirmware/Marlin #20598: [FR] PROBING_HEATERS_OFF - add option to keep the bed temperature

Sebazzz, 2020-12-29:
### Description

If I'm not wrong PROBING_HEATERS_OFF was introduced for the Creality CR-6 to mitigate electrical interference when reading the strain gauge. The strain gauge is read from the daughter board near the hot-end. The daughter board reads the analog signal from the strain gauge and passes that as digital signal (HIGH or LOW) to the motherboard where Marlin runs. There is no need to disable heating the bed while probing.

### Feature Workflow

This FR is to request to add an option to keep the bed heater on while probing and `PROBING_HEATERS_OFF` enabled.

1. `PROBING_HEATERS_OFF` enabled
2. Add option for keeping bed heater enabled
3. G28, G29 do not turn off the bed heater

#### Additional Information

Essentially [this](https://github.com/CR6Community/Marlin/commit/b1147ad54e6705389f2e4505aa45c48f73f38fca#diff-bb298d18dac4268225a7a53a5ad24844472530bdc63f6a98a2ba8d8158b75080R1091) plus [this](https://github.com/CR6Community/Marlin/commit/b1147ad54e6705389f2e4505aa45c48f73f38fca#diff-9c1d09e39e0d06d83cd4a44f54ff41e1cc364477c4dc39ecacf106a9b75540b7R2188).

thisiskeithb, 2020-12-29:
Wouldn't this be as simple as disabling `PROBING_HEATERS_OFF` in the CR-6 config? It appears to already be disabled in the config hosted in the Configurations repo at least.

Sebazzz, 2020-12-29:
That will keep the heater of the hot-end on, which is exactly the one you want to disable during probing.

sjasonsmith, 2020-12-29:
Could you simply probe before heating the nozzle?

Does heating the nozzle even interfere with probing? If I’m not mistaken the strain gauge is at the top of the hotend assembly, far from the nozzle.

Sebazzz, 2020-12-29:
No, because the nozzle is heated to make any filament remains soft and account for thermal expansion.

But the strain gauge emits an analogue signal. This is read by a HX711 chip on the daughter bord PCB but with the heaters on interference happens which causes an inaccurate read-out. 

Thats the entire point why that feature was added. The point is that it turns off the bed heater as well - but that isn't necessary nor desired.

thisiskeithb, 2020-12-29:
Until `PROBING_HEATERS_OFF` is split to allow the hotend & bed to be controlled independently, I'd go with something like this for start gcode (using PrusaSlicer's variables as an example):

```gcode
M104 S170 ; preheat hotend to 170
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S0 ; turn off hotend
G28 ; home all
G29 ; mesh bed leveling
```

sjasonsmith, 2020-12-29:
That probably isn’t good enough, since any ooze on the nozzle could re-harden and cause probe results to be too high.

Sebazzz, 2020-12-29:
That's a secondary issue actually: you only want the heater off in the probing zone (optical switch LOW). The nozzle can already be cold if you home from a large Z height.

huhansome, 2020-12-30:
阿里云服务器低至96/年，学生价9.5/年
https://www.aliyun.com/minisite/goods?source=5176.11533457&userCode=j6bryttg

阿里云数据库限时折扣，低至188/年
https://www.aliyun.com/activity/ambassador/database?source=5176.11533457&userCode=j6bryttg

阿里云企业级服务器1折限时特惠
https://promotion.aliyun.com/ntms/act/enterprise-discount.html?source=5176.11533457&userCode=j6bryttg

阿里云建站 速成美站 - 千套模板，价格低至500元
https://ac.aliyun.com/application/webdesign/sumei?source=5176.11533457&userCode=j6bryttg

CDN爆款产品低至5.5折
https://yqh.aliyun.com/live/cdncarnival?userCode=j6bryttg&source=5176.11533457&userCode=j6bryttg

thinkyhead, 2020-12-31:
> Does heating the nozzle even interfere with probing?

Heating in-general interferes with probe readings because the higher-voltage PWM tends to affect the electrical characteristics of the whole board and its attached peripherals. Disabling the heaters (just while polling the probe pin or awaiting a pin interrupt) removes this extra electrical interference and is demonstrated to produce more consistent readings with BLTouch, inductive probes, and capacitive probes.

thinkyhead, 2020-12-31:
If there is concern about heaters getting cold during a very long homing move, we could modify Marlin to leave the heaters on during the initial homing move, and then disable the heaters for the 'slow bump' homing move.

thisiskeithb, 2020-12-31:
> If there is concern about heaters getting cold during a very long homing move, we could modify Marlin to leave the heaters on during the initial homing move, and then disable the heaters for the 'slow bump' homing move.

How would that work with configs with no homing bump (sensorless homing)?

FanDjango, 2020-12-31:
> If there is concern about heaters getting cold during a very long homing move, we could modify Marlin to leave the heaters on during the initial homing move, and then disable the heaters for the 'slow bump' homing move.

The heating-turn-off is to early

thinkyhead, 2020-12-31:
> How would that work with configs with no homing bump (sensorless homing)?

If no homing bump was configured, then it would behave as it does now.

thinkyhead, 2020-12-31:
> The heating-turn-off is too early

Yes, that perfectly summarizes my point. The 'slow bump' move is only a millimeter or two long, so leaving the heaters on for the initial long homing move, and only turning them off before this short move should give the best result.

Sebazzz, 2020-12-31:
Does the last homing bump determine the actual zero position? Its not like the auto leveling by which the average of the two positions is taken?

Then it should probably be this,
1. Tare
2. Home & touch
3. Heaters off
4. Tare
5. Final homing move

FanDjango, 2020-12-31:
@Sebazzz Can you have a look at #20610?

Sebazzz, 2020-12-31:
> 
> 
> > If there is concern about heaters getting cold during a very long homing move, we could modify Marlin to leave the heaters on during the initial homing move, and then disable the heaters for the 'slow bump' homing move.
> 
> How would that work with configs with no homing bump (sensorless homing)?

Sensorless homing isn't used on the Z-axis, is it?

thisiskeithb, 2020-12-31:
> Sensorless homing isn't used on the Z-axis, is it?

it can be (`SENSORLESS_PROBING`), but not recommended. I was thinking of the homing bump for X/Y.

thinkyhead, 2021-01-05:
Marlin is now updated so that heaters will only be turned off for the second (slow) probe of the bed.

closed 2021-01-05 by thinkyhead