MarlinFirmware/Marlin #20603: Stopping printing causes the hotend to heat up

rkjames, 2020-12-30:
### Bug Description

Stopping printing causes the hotend to heat up. I expect it to go to 0F.

### Configuration Files

Attached. Also available at https://github.com/rkjames/Marlin/commit/0d9dcf45897b2378bf29c42b72fdab894f1054b6

### Steps to Reproduce

1. Start a simple print (CHEP_M0_bed_level.gcode, https://www.chepclub.com/bed-level.html)
2. Stop print

**Expected behavior:**

hot end cools down

**Actual behavior:**

hot end targets 200C

#### Additional Information

This reproduces consistently for me.

This did not happen before I flashed Marlin. Flash was bugfix-2.0.x (@4ad633bae235f0a47972d6d7873e614b8286d914) using Configurations\config\examples\Creality\Ender-3 V2 from Configuration@98ddd0ec8. (See https://github.com/rkjames/Marlin/commit/0d9dcf45897b2378bf29c42b72fdab894f1054b6)

Printer: Stock Ender 3v2

(I'm new to 3D printing and Marlin)

rkjames, 2020-12-30:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5755147/Configuration.zip)

sjasonsmith, 2020-12-31:
Are you printing from the SD card, or over serial?

rkjames, 2020-12-31:
SD card

slowbro, 2021-01-20:
@rkjames does this happen only with `CHEP_M0_bed_level.gcode`, or any gcode file you have printed?

closed 2021-02-27 by github-actions[bot]