MarlinFirmware/Marlin #20607: I know it's maybe late but i think there are still [perhaps more!] people having this issue so here is the **solution** in my case:

stroodies, 2020-12-31:
I know it's maybe late but i think there are still people having this issue so here is the **solution** in my case:

First my situation/problem:

1. The Printer works and u can move in X/Y Direction without any problems
2. When u press Auto Home (G28) the Printer moves to your home point, in my case the middle of the print bed.
3. The BlTouch lights red and the blue led is on
4. The BlTouch lights go off and it deploys -> stows the pin once.
5. Then u get the error + u cant move the Z Axis down, but in my case u can move it up

**Solution:**
In my case the Problem was due to no constant or no connection at all from the White/Black cable coming out from the BlTouch to the Printers Z-Endstop Pins. After making sure the connection there is perfect, i didnt get any errors and everything works fine.

Hope i could help anybody with this

_Originally posted by @FullMetalRico in https://github.com/MarlinFirmware/Marlin/issues/8926#issuecomment-437935571_

stroodies, 2020-12-31:
Hi buddies, I hope this is the right place for such a post-- just trying to save somebody a headache! 🥰

I had a very similar issue to this thread: https://github.com/MarlinFirmware/Marlin/issues/8926 
 except it wasn't wiring.

In my case, the solution was to comment out this line in configuration.h
//#define USE_ZMAX_PLUG
 
My issue was:
I was seemingly getting all the way through homing with BLtouch then getting an error:
Error: STOP called because of BLTouch error - restart with M999
Printer disconnects from host; either repetier or octoprint
The BLtouch was deploying correctly and reporting correctly when it touched the heat bed... which seems to cover full function...
But the printer crashed at the end of every homing cycle and had to be reset.

Solution:
It looks like leaving the Z_max_plug defined while I had the BLtouch's servo connections plugged in there was holding z_max in the triggered state.

I finally learned about the M119 command by reading lorenjz's post
M119 showed curious behavior from the endstops, especially the max endstops which I'm not using.
Turning  undefining USE_ZMAX_PLUG worked for me!

thisiskeithb, 2020-12-31:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-12-31 by thisiskeithb