MarlinFirmware/Marlin #20611: BLT v3.1 probing and reflecting grid but not actually compensating during print

WvZZ, 2020-12-31:
### Bug Description

Despite the BLT v3.1 probing and reflecting (i.e. using M503) a grid, there are no z commands during initial layer print as I would expect to adjust for the bed level and there is no visual movement in the z-axis rod (I put a piece of tape on it so make visual movements more noticeable) during a print (especially first layer).

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
Added "Github.zip"

If you've made any other modifications describe them in detail here.

I followed Teaching Teach's youtube video (https://www.youtube.com/watch?v=neS7lB7fCww&t=1173s) to compile the firmware. The only differences I used are that I use the probe for z stop.

### Steps to Reproduce

1. Slice file in Cura 4.8.0 (check that starting gcode has G28 and G29)
2. Switch on Ender 3V2
3. Open web to access Octoprint
4. Upload the above sliced gcode file
5. Select above uploaded gcode file
6. Begin print

**Expected behavior:**

I expect to see the BLT probing the bed and then I expect to see the z-axis move ever so slightly during (at least) the initial layer print. Using the bed visualizer plugin I can see that at least the plugin shows there will be z-axis movement.

**Actual behavior:**

BLT probes bed. Terminal shows a grid was formed and during the purge line I can see z commands in terminal and see the z-axis turn ever so slightly. This tells me I am reading the terminal commands correctly and can see even slight z-axis movements.  But, during the actual print there are no z commands (besides obviously layer increases) and no z-axis movement. 

I have included screenshots of (1) the terminal during the first layer - note no z-axis commands indicating z-axis compensation, (2) the report from a M503 during the print that reflects ABL is on (M420 S1) and that there is a grid and (3) the gcode being printed while the above two screenshots were taken. 

Other notes:
I have seen reports of a bug when using fade height. I have tried using fade off, fade height 4, fade height 10. All yield the same result. 

Other files:
[image]
[image]

[Spool_Bolt_long_(1)_nut_only.zip](https://github.com/MarlinFirmware/Marlin/files/5757162/Spool_Bolt_long_.1._nut_only.zip)

[Github.zip](https://github.com/MarlinFirmware/Marlin/files/5757153/Github.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

WvZZ, 2020-12-31:
Apologies - I am using the Bugfix 2.0.x branch

FanDjango, 2020-12-31:
When you are printing a layer, there won't be any movement commands for Z to be seen in the terminal, apart, as  you yourself already noted, for a layer change.

When the nozzle is moved from one X/Y position to the next, a Z adjustment is done "automagically" by Marlin. If you are not seeing that happening on the Z stepper motor, there are possible reasons for that.

Levelling could be turned off, its not using the mesh. You said that you can see some levelling taking place on the purge line. If that is true, I would look for the problem in the Start GCODE of your sliced object.

FanDjango, 2020-12-31:
An additional thing you can test is to do some long distance `GO Xnnnn Ynnnn` commands yourself first.

WvZZ, 2020-12-31:
Hi, FanDjango.

During the Purge line there always was movement - I think Cura does that to clear the nozzle. 

This is my start Gcode in Cura 4.8.0:

`; Ender 3 Custom Start G-code

G92 E0 ; Reset extruder
M190 S{material_bed_temperature} ; start heating bed to what is set in Cura AND wait before bed leveling
M290 P0 S160 ; BLtouch alarm release
G4 P100 ; delay for BLtouch
G28 ; Home
G29 ;ABL

; To use ABL to level in this print, comment out M501 and M420 S1 
; and uncomment G20 
;M501
;M420 S1

M190 S{material_bed_temperature} ; start heating bed to what is set in Cura AND wait
M109 S{material_print_temperature} T0 ; start heating hotend to what is set in Cura and WAIT
G92 E0 ; Reset Extruder`

I've also tried commenting out "G29" and uncommenting "M501" and "M420 S1"

From the mid-print M503 report, I can see ABL is enabled (M420 S1) and there is a grid. 

Anything else I should look at?

FanDjango, 2020-12-31:
I can see you use  `#define ENABLE_LEVELING_AFTER_G28`. Thats fine.

I don`t see the purge line in your start gcode? Who is doing it? Octoprint setting?

To be safe I always also put the `M420 S1` after the G29.

FanDjango, 2020-12-31:
Here is mine:
```
M502        ; Set to Marlin defaults
M500        ; Store in EEPROM

M107 P0     ; Part cooling fan off
M107 P1     ; Extruder fan off

; turn on bed heating with 'nowait' to save some time
M140 S{material_bed_temperature_layer_0} ; nowait

; turn on nozzle heating to 165° to save some time
M104 S165  ; nowait

G90        ; Absolute

G28        ; Home

G34        ; Precision align Z

G1 X180 Y{machine_depth} Z100 F900 ; Present bed for cleaning

; now we need the bed at print temperature
M190 S{material_bed_temperature_layer_0} ;wait

; bed levelling
G29
M420 S1      ; set Leveling fade height to 10mm (default)

; now start heating nozzle to final temperature
M104 S{material_print_temperature_layer_0, 0} ;nowait

G1 X180 Y0 Z100 F900 ; present nozzle for cleaning

; now wait for both final temperatures to be reached
M109 S{material_print_temperature_layer_0, 0} ;wait
M190 S{material_bed_temperature_layer_0}      ;wait

; wipe the nozzle at the front of the bed to prime it
M83                     ; E relative
G92 ﻿E0.0                ; reset extrusion distance
G0 X10 Y5 Z0.3   F9000  ; prime start pos
G0 Z0.3           F900

G1 X30     E+6.0  F600  ; 30mm intro line
G1 X70     E+1.5  F600  ; +40mm intro line
G1 X110    E+3.0  F600  ; +40mm intro line

G1         E-0.5 F1800  ; retract to avoid stringing

G1 X116    E0     F600  ; +6mm intro line
G1 X111    E0     F600  ; -5mm intro line

G0 Z10            F900  ; Up to 10mm
G0 X180          F9000  

G92 E0.0                ; reset extrusion distance

M84 S600   ; steppers off if inactive longer than 10 minutes (safety)

M900 K.12  ; Linear Advance Coefficient (experimental)

M82        ; E absolute (currently, this is absolutely expected by CURA
```

WvZZ, 2020-12-31:
Here is my full start gcode in Cura. It's long as I have that pre-fix for Octolape:

`; Script based on an original created by tjjfvi (https://github.com/tjjfvi)
; An up-to-date version of the tjjfvi's original script can be found
; here:  https://csi.t6.fyi/
; Note - This script will only work in Cura V4.2 and above!
; --- Global Settings
; layer_height = {layer_height}
; smooth_spiralized_contours = {smooth_spiralized_contours}
; magic_mesh_surface_mode = {magic_mesh_surface_mode}
; machine_extruder_count = {machine_extruder_count}
; --- Single Extruder Settings
; speed_z_hop = {speed_z_hop}
; retraction_amount = {retraction_amount}
; retraction_hop = {retraction_hop}
; retraction_hop_enabled = {retraction_hop_enabled}
; retraction_enable = {retraction_enable}
; retraction_speed = {retraction_speed}
; retraction_retract_speed = {retraction_retract_speed}
; retraction_prime_speed = {retraction_prime_speed}
; speed_travel = {speed_travel}

; Ender 3 Custom Start G-code

G92 E0 ; Reset extruder
M190 S{material_bed_temperature} ; start heating bed to what is set in Cura AND wait before bed leveling
M290 P0 S160 ; BLtouch alarm release
G4 P100 ; delay for BLtouch
G28 ; Home
G29 ;ABL

; To use ABL to level in this print, comment out M501 and M420 S1 
; and uncomment G20 
;M501
;M420 S1

M190 S{material_bed_temperature} ; start heating bed to what is set in Cura AND wait
M109 S{material_print_temperature} T0 ; start heating hotend to what is set in Cura and WAIT
G92 E0 ; Reset Extruder

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish`

I thought let me also try printing from the SD card to eliminate a possible Octprint issue. I still don't see any z-axis movement and the printer did do the probing first.

I don't think its my code as I've also tried mid-print to enter "M420 S1" into terminal and still no luck. 

But, after the current print, I'll try adding a "M420 S1" immediately after G29 to see if that helps. 

I also saw somewhere (but didn't save the page) that there is a M34 or M43 or something you can include with/after M420... Do you know what I'm talking about? SHould I try adding anything extra to my G29 command?

(P.s Thank you so much for your help. I sincerely appreciate it! I know you are giving freely of your time and just know I am grateful!)

FanDjango, 2020-12-31:
Now I can see the purge line. :-)
> But, after the current print, I'll try adding a "M420 S1" immediately after G29 to see if that helps.

Actually it shouldn't help, if everything is ok. We will see.

FanDjango, 2020-12-31:
I just did the following, using only the LCD:

```
Auto-Home
Level Bed
```
note: Below that, is an indicator of the Levelling status. While G29 is running it will be "off", note also, Level Bed (=G29) will do another home when it is finished. Then the status will be "on"

Then I went into the Motion menü and moved the Z axis down to about 1.0mm height (so that fade height will be not too active) and then moved X and Y over long distances. I could see the Z steppers moving, on my printer about 5 degrees. Not much.

You really need to do more testing, not just printing. You could extract the start gcode from your object with an editor and put two `G0 Xnnn Ynnn` long-distanct commands in there as a "dummy" print and see if the Z stepper moves. Stuff like that. 

Consider moving this to Discord in the support channel. I don't think there is a bug to be found.

WvZZ, 2020-12-31:
Thanks, FanDjango. 

I added a msg to the Marlin Support channel under my name "CompostBox". Hopefully I can sort this out.

Have an awesome day! (Happy New Year!)

thisiskeithb, 2020-12-31:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

WvZZ, 2021-01-02:
Hi, ThisIsKeithB

Thanks - I'll enable those features and upload once I've had a chance to test it out. 

I will also purposefully tilt my bed to exaggerate any movements and help visually see what is happening. Please give me a few days to test and I'll get back to this thread. 

Thank you. Happy new year :)

WvZZ, 2021-01-03:
Hi, ThisIsKeithB

I reflashed the firmware with those options enabled and sent the M11 S247 commands to the printer via terminal in Octoprint and ran a print last night. Do you know where I can find the output file? Or if I'm using Octoprint do I need to first enable serial logging or install a plugin first? Sorry, I thought the output would simply be under "logging" in Octoprint but I don't see it there. There is a normal log file but it doesn't have much and it all looks related to Octoprint and not what the printer is doing. 

Thank you. 

Thank you.

WvZZ, 2021-01-03:
Hi, ThisIsKeithB

I enabled serial logging in Octprint and it worked. This time, i just fusion-360'd a curvey line that would cover quite a bit of distance in the X and Y plane to exaggerate any z-axis compensation. I also specifically tilted my bed towards the bottom left. It now seems like the ABL is working. I suspect ABL only 'kicks in' after a certain level of compensation is required. 

I've attached a zip folder with the following in it:
> serial log file
>  video showing z-axis moving
> Picture of final print showing relatively good adhesion despite being at most 0.5 out
> the STL I printed
>The gcode from the Cura 4.8.0 slicer

Do you agree and see that ABL is working as it should? 

Do you know whether ABL only kicks in at a certain level and, if so, what that level is? 

Thank you for your help, ThisIsKeithB. I sincerely appreciate it!

[Debugging.zip](https://github.com/MarlinFirmware/Marlin/files/5761246/Debugging.zip)

WvZZ, 2021-01-03:
Quick question, is there any problem with me leaving the DEBUG_LEVELING_FEATURE and M114_DETAIL firmware version in my printer or is it better to reflash to a version without DEBUG_LEVELING_FEATURE and M114_DETAIL?

Thanks.

Talha909, 2021-01-11:
I have same kind of problem. I have probe my bed with 9 points. Probing  was successful. But when the print starts. It print perfectly on front side. But when it goes back nozzle comes very near to the bed and filament has not way to come out from the nozel. But it comes from back to front . It start again coming from the nozeel. from front half the print is very fine and in end half there is lot of un even surface and thin layer. And in corner lot of filament.  Whats is the reason

thestephenmarshall, 2021-01-13:
@Talha909 if you run the "Mesh Validation Sequence" as seen [here](https://marlinfw.org/docs/gcode/G026.html), do you get the same results? You will need `#define G26_MESH_VALIDATION` un-commented for this to work. Here's an example where my nozzle is `260C`, my bed is `100C` and my desired layer height is `0.2` : `G26 B100 H260 L0.2`

closed 2021-02-20 by github-actions[bot]