MarlinFirmware/Marlin #20612: _USB_HOST

BuckBREW, 2020-12-31:
https://github.com/KAY3D/Marlin/blob/bdd3b369a953ca4b746cf5352178ecf54ed565b2/Marlin/src/sd/usb_flashdrive/lib-uhs2/settings.h#L145-L145

ellensp, 2020-12-31:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2020-12-31:
We can’t support forks of Marlin here. You’ll need to submit your issue to KAY3D’s fork.

closed 2020-12-31 by thisiskeithb