MarlinFirmware/Marlin #20614: [ABL] (ABL leveling does ony compensate half of it)

kampfhase7, 2020-12-31:
### Bug Description

I have changed the config for the orginal creality BL-Touch V3.1. All is working fine. But in the print after leveling via G29, the Z axix is moving only a half of the vaules from the BL-Touch. 
Example: BL-Touch probing the bed. X-10 to X+10 it haves a height difference of 1mm. While Printing, Z only moves this way only 0.5mm. But have to moving 1mm up. Im using Bugfix branche on my Ender 3v2 with board v4.2.2. 

Im using bilinear leveling

### Configuration Files
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5757488/configs.zip)

FanDjango, 2020-12-31:
You have `#define ENABLE_LEVELING_FADE_HEIGHT`. Check that the default of 10 isn't changed anywhere in the start gcode. That could cause less correction at a low height.

thisiskeithb, 2020-12-31:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

thestephenmarshall, 2021-01-07:
Check that your motor currents for each stepper are identical. https://marlinfw.org/docs/gcode/M906.html

e.g. `M906 I1 Z800` + `M906 Z800`.

domhardt, 2021-01-10:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
>
> * Download Marlin [...]

I configured and compiled the latest bugfix version and followed the instructions. Please find the log file and my configuration files attached. The printer still does not fully compensate for the offset measured by the BLTouch.

[Log+Config.zip](https://github.com/MarlinFirmware/Marlin/files/5793403/Log%2BConfig.zip)

thestephenmarshall, 2021-01-11:
**Mesh validation prints perfectly...**

I've been chasing this issue for a year now with some resolve only to have it pop right back up again. SKR v1.3 + TMC2209 drivers. I've tweaked everything in the firmware, blueprinted everything in the hardware and still out of nowhere the bug is back again. Let me just leave this information here: 

- Billinear leveling
- 3 x 3 grid
- 3 point bed and carriage mounts
- BL Touch v3.1
- `G26 H260 B100 L0.2` - mesh validation prints perfectly
- Print from the attached gcode sliced in Cura or Prusa Slicer - left side is too high every single time
- Manually adjust the the mesh `M421` on the left side - no difference (how??) - yes, I saved the mesh
- Yes, I have `M420 S1 Z10` at the start of my prints
- Octoprint latest on Raspberry Pi 4
- Print from SD - no difference, too high on left side
- `G34` - 0.0 difference left to right
- Probe repeatability test so many times I need a tattoo now - it passes every time

Has anyone ran the mesh validation command from the above list (using your own temperatures of course) and found the result to be perfect yet any gcode you print (please try mine) is borked? 

I don't know a lot about the languages that compose this firmware, but can anyone point me to where I can start to learn how ABL works? What's a good starting point? I feel like, from all of my testing, ABL is simply not compensating correctly even though it is most definitely working as I marked the lead screw couplers with a chalk pen and can see them moving as I would expect. Yes, I also did check those for snugness and well and most certainly cannot move the lead screws inside of their couplers. 

As a workaround, I have printed the attached calibration circle (100mm) in the center of the bed and simply paused the print then turned off the motors to manually adjust the left side down enough to make a decent first layer. This is a hack and I do no like it. 

**Why is it that I can run the G26 mesh validation and get perfect results but garbage results when I run the attached GCODE file?**

[image]

GCODE:
[100mm-cal-circle.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/5797670/100mm-cal-circle.gcode.zip)

NastyEwok, 2021-02-17:
> **Mesh validation prints perfectly...**
> 
> I've been chasing this issue for a year now with some resolve only to have it pop right back up again. SKR v1.3 + TMC2209 drivers. I've tweaked everything in the firmware, blueprinted everything in the hardware and still out of nowhere the bug is back again. Let me just leave this information here:
> 
> * Billinear leveling
> * 3 x 3 grid
> * 3 point bed and carriage mounts
> * BL Touch v3.1
> * `G26 H260 B100 L0.2` - mesh validation prints perfectly
> * Print from the attached gcode sliced in Cura or Prusa Slicer - left side is too high every single time
> * Manually adjust the the mesh `M421` on the left side - no difference (how??) - yes, I saved the mesh
> * Yes, I have `M420 S1 Z10` at the start of my prints
> * Octoprint latest on Raspberry Pi 4
> * Print from SD - no difference, too high on left side
> * `G34` - 0.0 difference left to right
> * Probe repeatability test so many times I need a tattoo now - it passes every time
> 
> Has anyone ran the mesh validation command from the above list (using your own temperatures of course) and found the result to be perfect yet any gcode you print (please try mine) is borked?
> 
> I don't know a lot about the languages that compose this firmware, but can anyone point me to where I can start to learn how ABL works? What's a good starting point? I feel like, from all of my testing, ABL is simply not compensating correctly even though it is most definitely working as I marked the lead screw couplers with a chalk pen and can see them moving as I would expect. Yes, I also did check those for snugness and well and most certainly cannot move the lead screws inside of their couplers.
> 
> As a workaround, I have printed the attached calibration circle (100mm) in the center of the bed and simply paused the print then turned off the motors to manually adjust the left side down enough to make a decent first layer. This is a hack and I do not like it.
> 
> **Why is it that I can run the G26 mesh validation and get perfect results but garbage results when I run the attached GCODE file?**
> 
> [image]
> 
> GCODE:
> [100mm-cal-circle.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/5797670/100mm-cal-circle.gcode.zip)

I seem to be experiencing similar issues.

Ill get some data together and post it

thestephenmarshall, 2021-04-08:
Let's not close this. The issue persists with a COMPLETELY NEW machine using an inductive probe and spring steel plate. I'm convinced the issue is either the LPC1768 SKR v1.3 hardware or Marlin's handling around said hardware (other hardware as well?). I am in no position to fix anything within this repo as it is not my area of coding expertise, but I will provide logs if need be despite the fact that multiple issues have been open on this topic. I have yet to see any convincing results from others who have provided logs within this or other issues. That being said, again, I would love to know why `G26` produces perfect ABL results while prints which enable this starting GCODE:

```
G28;
G29;
M500;
M501;
M420 S1 Z1;
```

exhibit the *exact same results described in this and other tickets*. Thoughts?

hamster65, 2021-04-08:
Does your printer do anything different when running gcode and mesh validation? Different motor currents, homing procedure, more/less faster/slower z movements?

To rule out a couple of things, you could try
- A different slicer
- disable fade height
- swith on / off interpolation
- enable / disable extrapolate beyond grid
- enable / disable subdivision
- UBL instead of ABL

I had better results with UBL, tested Klipper (levelling worked) and switched back to Marlinßßß. Testing will continue once I have access to my printer again and more time.

thestephenmarshall, 2021-04-13:
@hamster65 when I try to enable UBL I get only 1 point then it says it's done. I will try again bc iirc UBL worked much better than billinear.

thestephenmarshall, 2021-04-14:
Figured that out, but with UBL now the problem side is the back instead of the right side of the bed. 🤷‍♂️

Using mesh bed leveling as a workaround. Can't spend any more time on this unfortunately. Best of luck.

GdLakeIoT, 2021-04-27:
Not sure this is the same, but I have a version of Bug-Fix2.x that will correctly compensate bead leveling. The version I have is from Oct 2 2020

closed 2022-04-21 by thinkyhead