MarlinFirmware/Marlin #20615: [FR] Custom movement speed during G26 validation pattern

ScrewThisBanana, 2020-12-31:
### Description

I would like to set the G26 movement speed to a value different from the G26 extrusion Speed

### Feature Workflow

1. Configure the `G26_XY_FEEDRATE_MOVE` parameter
2. Software check if extrusion takes place, if not, `G26_XY_FEEDRATE_MOVE` is used instead of `G26_XY_FEEDRATE`

#### Additional Information

Possible implementation in https://github.com/ScrewThisBanana/Marlin/tree/bugfix-2.0.x

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

qwewer0, 2021-01-01:
@ScrewThisBanana Could you make a PR on the main bugfix-2.0.x branch?

ScrewThisBanana, 2021-01-09:
@qwewer0 Will test it on my printer once more and open a pull request after that.

ScrewThisBanana, 2021-01-09:
> @ScrewThisBanana Could you make a PR on the main bugfix-2.0.x branch?

Done. PR is ready for merge

qwewer0, 2021-02-25:
I think this can be closed.

ScrewThisBanana, 2021-02-25:
Closed - merged.

closed 2021-02-25 by ScrewThisBanana