MarlinFirmware/Marlin #20618: [BUG] HOMING_FEEDRATE_XY was not declared in this scope

reallymello, 2020-12-31:


sjasonsmith, 2020-12-31:
This is a mismatch between Marlin and example versions.

Your Marlin source is not the latest `bugfix-2.0.x` source. The `HOMING_FEEDRATE_XY` and `HOMING_FEEDRATE_Z` constants no longer exist anywhere in the Marlin codebase.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-31 by sjasonsmith

sjasonsmith, 2020-12-31:
To correct my earlier statement: They do still exist, but only in SanityChecks.h to tell you that they should no longer be used.