MarlinFirmware/Marlin #20623: [BUG] 'HOMING_FEEDRATE_XY' was not declared in this scope

JRitmeester, 2021-01-01:
Downloaded Marlin 2.0.7.2 files to update the firmware on my Ender 3.

```Arduino: 1.8.13 (Windows 10), Board: "Sanguino, ATmega1284 or ATmega1284P (16 MHz)"

In file included from c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\inc\marlinconfig.h:47:0,

                 from sketch\src\module\motion.h:31,

                 from sketch\src\module\motion.cpp:27:

sketch\src\module\motion.cpp:151:16: error: 'HOMING_FEEDRATE_XY' was not declared in this scope

     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),

                ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

sketch\src\module\motion.cpp:151:16: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'

     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),

                ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

sketch\src\module\motion.cpp:151:48: error: 'HOMING_FEEDRATE_XY' was not declared in this scope

     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),

                                                ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

sketch\src\module\motion.cpp:151:48: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'

     MMM_TO_MMS(HOMING_FEEDRATE_XY), MMM_TO_MMS(HOMING_FEEDRATE_XY),

                                                ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

sketch\src\module\motion.cpp:153:14: error: 'HOMING_FEEDRATE_Z' was not declared in this scope

   MMM_TO_MMS(HOMING_FEEDRATE_Z)

              ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

sketch\src\module\motion.cpp:153:14: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'

   MMM_TO_MMS(HOMING_FEEDRATE_Z)

              ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

sketch\src\module\motion.cpp: In function 'void do_z_clearance(const float&, bool, bool, bool)':

sketch\src\module\motion.cpp:513:100: error: 'HOMING_FEEDRATE_Z' was not declared in this scope

   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));

                                                                                                    ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:196:29: note: in expansion of macro 'THIRD'

 #define ___TERN(P,V...)     THIRD(P,V)              // If first argument has a comma, A. Else B.

                             ^~~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:195:29: note: in expansion of macro '___TERN'

 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'

                             ^~~~~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:194:29: note: in expansion of macro '__TERN'

 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'

                             ^~~~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:190:29: note: in expansion of macro '_TERN'

 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'

                             ^~~~~

sketch\src\module\motion.cpp:513:60: note: in expansion of macro 'TERN'

   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));

                                                            ^~~~

sketch\src\module\motion.cpp:513:100: note: suggested alternative: 'HOMING_FEEDRATE_MM_M'

   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));

                                                                                                    ^

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\types.h:76:43: note: in definition of macro 'MMM_TO_MMS'

 #define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)

                                           ^~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:196:29: note: in expansion of macro 'THIRD'

 #define ___TERN(P,V...)     THIRD(P,V)              // If first argument has a comma, A. Else B.

                             ^~~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:195:29: note: in expansion of macro '___TERN'

 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'

                             ^~~~~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:194:29: note: in expansion of macro '__TERN'

 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'

                             ^~~~~~

c:\users\jeroen\appdata\local\temp\arduino_build_81492\sketch\src\core\macros.h:190:29: note: in expansion of macro '_TERN'

 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'

                             ^~~~~

sketch\src\module\motion.cpp:513:60: note: in expansion of macro 'TERN'

   do_blocking_move_to_z(_MIN(zdest, Z_MAX_POS), MMM_TO_MMS(TERN(HAS_BED_PROBE, Z_PROBE_SPEED_FAST, HOMING_FEEDRATE_Z)));

                                                            ^~~~

exit status 1

Error compiling for board Sanguino.```

Didn't change any code except the Bootscreen bitmap.

ellensp, 2021-01-01:
You are using the wrong config files. You have to use the examples from the release-2.0.7.2 branch not the default import branch ie from here https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2/config/examples/Creality/Ender-3

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-01-01 by ellensp

ellensp, 2021-01-01:
Ie this fie https://github.com/MarlinFirmware/Configurations/archive/release-2.0.7.2.zip

JRitmeester, 2021-01-01:
Then I was misguided by the tutorial I was following by [Crosslink](https://www.youtube.com/watch?v=RbbzsJBpEhc) (adding the link in case anyone else comes across this issue), since he tells you to get the files from [this page](https://github.com/MarlinFirmware/Configurations). Did I miss something?

thisiskeithb, 2021-01-01:
> Then I was misguided by the tutorial I was following by Crosslink (adding the link in case anyone else comes across this issue), since he tells you to get the files from this page. Did I miss something?

The information was incorrect. Marlin is not affiliated with that YouTuber, so we can't control information they put out.

It's best to stick to https://marlinfw.org/meta/download/ unless you know what you're looking for.