MarlinFirmware/Marlin #20625: [HELP needed] SD Init Fail on custom RRD smart controller + BTT SKR e3 mini v2 "adapter"

wikolii, 2021-01-01:
### Bug Description

First of all, i don't consider this a BUG or a Feature request, so here goes:
I'm looking for HELP hooking up my REPRAP_DISCOUNT_SMART_CONTROLLER to my BIGTREETECH-SKR-mini-E3 v2.
I got it working on some printers but i always find a problem with reading the SD card correctly, i'm thinking there's a random issue with trying to do this, because the "SD Init Fail" error goes away after many tries, never returns but i don't know what i'm doing to mitigate it. 
So the actual question is: **Does anybody know _which pin/pins_ could trigger this behavior?** SD_DETECT_PIN works and LCD_SDSS is my mayor suspect, but when i test conductivity i see no fails.
What actually happens when checking the SD Init process?

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Config files + PINS.zip](https://github.com/MarlinFirmware/Marlin/files/5759325/Config.files.%2B.PINS.zip)

I'm including both config files and the pins files i changed in order to make them work.
I'm using an inductive sensor that i had to move to a different pin than Z_STOP_PIN because i was getting always triggered, i guess there's a resistor on the board that was causing that. So i assume, it's kind of messy but i got it working on 4 printers already.

### Steps to Reproduce

1. Power on
2. Insert SD card on LCD 

**Expected behavior:**

Recognize the SD and select files as i normally do

**Actual behavior:**

I get "SD Init Fail" error

#### Additional Information

This photos are only to get the idea of the how my new "adapter" works.
[image]
[image]

**Clarifications:**

- I know this board is created for ender machines. Mine are custom made prusa bear-like printers.
- I know i can get LCD controllers that are plug and play with this board. But they cost money and i have to upgrade 11 printers. Plus i already have all the RRD smart controllers i need.
- I already got this working on 4 printers so far, i know that eventually it will start working on the one that i'm stuck with now. But maybe someone comes up with their similar experience and how they solved a similar issue.
- I already tried with 10 different SD cards that already work on the same setup on other machines.
- I got into this from an Arduino Mega + RAMPS 1.4 + DRV8825 setup. I wanted to go quiet and upgrade to 32 bits.
- In retrospect maybe i should have gone with SKR 1.4 + TMC2209. The money was not much more and that board has the 2 EXP connectors for this smart controller. But i already bought all the mini boards i need for all the printers and can't return them.
- I live in argentina, here we can't get anything in a couple of days for the right price. Here we have to make due with what we can get and spend as little as we can. So we make up with manual work, like making our own adapters.

Thanks in advance, have a great 2021!

rhapsodyv, 2021-01-31:
did you try lowering sd spi speed?
```
#define SD_SPI_SPEED SPI_EIGHTH_SPEED
```

wikolii, 2021-02-05:
I will try when i get back home. Thanks!

wikolii, 2021-03-10:
> did you try lowering sd spi speed?
> 
> ```
> #define SD_SPI_SPEED SPI_EIGHTH_SPEED
> ```

This worked, thanks!

closed 2021-03-10 by wikolii